micropython/micropython #2713: pyboard UART reads a spurious character

peterhinch, 2016-12-21:
Link X1 and X2. Paste the following:
```python
from pyb import UART
uart = UART(4, baudrate = 9600)
txt = b'A line of text.'
nchars = uart.write(txt)
print('Wrote {} chars. Len = {}'.format(nchars, len(txt))) # correct
rec = uart.read() # Returns b'A line of text..'
print('Got {}. Len = {}'.format(rec, len(rec)))
```
Outcome:
```
MicroPython v1.8.1-925-g46e59c5-dirty on 2016-12-13; PYBv1.1 with STM32F405RG
Type "help()" for more information.
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
[paste omitted]
Wrote 15 chars. Len = 15
Got b'A line of text..'. Len = 16
>>> 
```
The last character is repeated in uart.read().

dpgeorge, 2016-12-22:
Thanks for the report, I can confirm the bug.  It's actually sending the last character twice (in some circumstances).

The issue is that the latest update to the HAL changed the semantics of the HAL's timeout (at least for the UART transmit function).  Before the change, the timeout was per character.  With the latest HAL the timeout is for the entier function, ie for all characters.  So the timeout is expiring after sending about 4-5 chars.

But then uPy notices it is a short write and tries to write the remaining chars.  But the calculation for how many chars were actually sent before the timeout expired is incorrect if the timeout occurs for the last character sent.  And so the last char is sometimes resent.

Because the HAL doesn't provide a robust way of telling how many chars were sent before a timeout, the only way to fix this problem properly is to not use the HAL and instead write a custom UART tx function.  I have something along these lines that seems to work (for F4, F7, L4) but it needs some tidying up before committing.

chuckbook, 2016-12-22:
In my opinion recent HAL_UART_Transmit is definitely buggy, at least for f4.
Would it make sense to have a machine.UART for stmhal? If yes, which level of HA would be ok?

dpgeorge, 2016-12-23:
> Would it make sense to have a machine.UART for stmhal?

The existing pyb.UART needs to be fixed, at the very least.  Then machine.UART can link to pyb.UART because they both have the same semantics for the core methods.

> which level of HA would be ok?

I opted to implement the UART tx function from scratch, because it's not that difficult.  See #2717.

chuckbook, 2016-12-23:
Great! Now it might be possible to have an option that doesn't wait for frame completion. Think of very low baud rates (1200 or even less). Even a distinct error flag for final timeout would be helpful.

peterhinch, 2016-12-23:
Great work - and fast! As far as I'm concerned the issue can be closed.

peterhinch, 2016-12-27:
This bug has reappeared: same hardware, updated firmware.
```
PYB: sync filesystems
PYB: soft reboot
MicroPython v1.8.1-1042-g05aebb9-dirty on 2016-12-25; PYBv1.1 with STM32F405RG
Type "help()" for more information.
>>> 
[paste of same code omitted]
Wrote 15 chars. Len = 15
Got b'A line of text..'. Len = 16
>>> 
```

chuckbook, 2016-12-27:
I tried the example from @peterhinch with random baudrates (1200..2400000) and couldn't reproduce the trailing char bug. However, it looks like for 13000 < baudrates < 19200 the minimum character timeout is too short. This makes sense as systick (granularity 1ms) is used as timebase.
I propose to use a timeout bigger than **(12*1000)/baudrate+2** to be on the safe side.

dpgeorge, 2016-12-28:
> This bug has reappeared: same hardware, updated firmware.

The patch was only just merged, so probably you used firmware that didn't have the fix.  Please report if you still have the bug.

> I propose to use a timeout bigger than (12*1000)/baudrate+2 to be on the safe side.

Timeout value has been increased by 1ms in e81116d07d3fed43f77f81bd5e20800d9d798f5f

> it might be possible to have an option that doesn't wait for frame completion

That would be nice, but I'm not quite sure how to do it.  Probably need to wait/check for frame completion at the start of the tx function (as opposed to the end), and then raise some kind of error if it doesn't complete in time.  Feel free to suggest a patch for this :)

chuckbook, 2016-12-28:
@dpgeorge , thanks for putting this in. Regarding the trailing frame completion I used the optimistic strategy and deleted the trailing UART_FLAG_TC check in uart_tx_data. I thought of using negative timeout_char value to turn this feature off, but this would be just a hack :-)

dpgeorge, 2017-01-17:
The issue with extra chars is now fixed.

closed 2017-01-17 by dpgeorge