micropython/micropython #2714: isprintable

dmazzella, 2016-12-21:
I tried to implement str.isprintable for increase coverage but i have found an incongruence:

in [unicode.c](https://github.com/micropython/micropython/blob/master/py/unicode.c#L51)  the table report 0, AT_SP, AT_SP, AT_SP, AT_SP, AT_SP, 0, 0, for this reason the result is:

```
Python 3.4.3 (default, Mar  4 2016, 10:23:36)
[GCC 4.9.2] on msys
Type "help", "copyright", "credits" or "license" for more information.
>>> [print(c, chr(c).isprintable()) for c in range(128)]
0 False
1 False
2 False
3 False
4 False
5 False
6 False
7 False
8 False
9 False
10 False
11 False
12 False
13 False
14 False
15 False
16 False
...
```

```
MicroPython v1.8.6-201-gbb688b2-dirty on 2016-12-21; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> [print(c, chr(c).isprintable()) for c in range(128)]
0 False
1 False
2 False
3 False
4 False
5 False
6 False
7 False
8 False
9 True
10 True
11 True
12 True
13 True
14 False
15 False
16 False
...
```

any reason for this?

dmazzella, 2016-12-21:
the fix is remove in [unicode.c](https://github.com/micropython/micropython/blob/master/py/unicode.c#L42) ` | FL_PRINT`

dmazzella, 2016-12-21:
the correct table can be:

```
// shorthand character attributes
#define AT_PR (FL_PRINT)
#define AT_SP (FL_SPACE)
#define AT_PS (FL_PRINT | FL_SPACE)
#define AT_DI (FL_DIGIT | FL_PRINT | FL_XDIGIT)
#define AT_AL (FL_ALPHA | FL_PRINT)
#define AT_UP (FL_UPPER | FL_ALPHA | FL_PRINT)
#define AT_LO (FL_LOWER | FL_ALPHA | FL_PRINT)
#define AT_UX (FL_UPPER | FL_ALPHA | FL_PRINT | FL_XDIGIT)
#define AT_LX (FL_LOWER | FL_ALPHA | FL_PRINT | FL_XDIGIT)

// table of attributes for ascii characters
STATIC const uint8_t attr[] = {
    0, 0, 0, 0, 0, 0, 0, 0,
    0, AT_SP, AT_SP, AT_SP, AT_SP, AT_SP, 0, 0,
    0, 0, 0, 0, 0, 0, 0, 0,
    0, 0, 0, 0, AT_SP, AT_SP, AT_SP, AT_SP,
    AT_PS, AT_PR, AT_PR, AT_PR, AT_PR, AT_PR, AT_PR, AT_PR,
    AT_PR, AT_PR, AT_PR, AT_PR, AT_PR, AT_PR, AT_PR, AT_PR,
    AT_DI, AT_DI, AT_DI, AT_DI, AT_DI, AT_DI, AT_DI, AT_DI,
    AT_DI, AT_DI, AT_PR, AT_PR, AT_PR, AT_PR, AT_PR, AT_PR,
    AT_PR, AT_UX, AT_UX, AT_UX, AT_UX, AT_UX, AT_UX, AT_UP,
    AT_UP, AT_UP, AT_UP, AT_UP, AT_UP, AT_UP, AT_UP, AT_UP,
    AT_UP, AT_UP, AT_UP, AT_UP, AT_UP, AT_UP, AT_UP, AT_UP,
    AT_UP, AT_UP, AT_UP, AT_PR, AT_PR, AT_PR, AT_PR, AT_PR,
    AT_PR, AT_LX, AT_LX, AT_LX, AT_LX, AT_LX, AT_LX, AT_LO,
    AT_LO, AT_LO, AT_LO, AT_LO, AT_LO, AT_LO, AT_LO, AT_LO,
    AT_LO, AT_LO, AT_LO, AT_LO, AT_LO, AT_LO, AT_LO, AT_LO,
    AT_LO, AT_LO, AT_LO, AT_PR, AT_PR, AT_PR, AT_PR, 0
};
```
Added new define AT_PS used for char at position 32.

With this modification the result with this script are identical:

`_ = [list(map(lambda x: print(x, ':', c, '=', getattr(chr(c), x)()), ['isspace', 'isalpha', 'isdigit', 'isupper', 'islower', 'isprintable'])) for c in range(128)]`

dmazzella, 2016-12-21:
for test is use this [repo](https://github.com/dmazzella/micropython/tree/string_istest)

dpgeorge, 2016-12-22:
Thanks for looking into coverage testing.

The function `unichar_isprint` is actually never used (especially since c305ae32436e37327d33df979d57d9ac1fb822c1) so I think the simplest thing is to just remove it (or comment it out, in case it's needed in the future).

closed 2016-12-22 by dmazzella