micropython/micropython #2716: Optimize iteration over empty tuple/list

pfalcon, 2016-12-22:
It appears that code like:

~~~
for i in ():
~~~

leads to unconditional allocation of (empty) iterator. It might make sense to check for a case of empty known-length sequence, and return some static "empty" iterator instead.

pfalcon, 2016-12-22:
This came up when investigating why uasyncio again allocated memory while it shouldn't.

dpgeorge, 2016-12-23:
A solution that covers this case as well as most non-emtpy iterations is #2416.

pfalcon, 2016-12-23:
Ok, the simple solution would be to just check for empty sequence in each ->getiter() and return special static empty iterator, but that would only add more code to later add more code with #2416. So, let's concentrate on merging the latter.

dpgeorge, 2016-12-28:
A workaround for this issue (that may or may not work, depending on the iterator) is to iterate using a range and then index the iterator, eg:
```
for index in range(len(iterator)):
    i = iterator[index]
    ...
```

pfalcon, 2016-12-28:
The simplest (in some sense) workaround is to add a test for empty sequence before iterating it: https://github.com/micropython/micropython-lib/commit/f29be360c0d834fc8638ee5b9aa6d1ef22c89093