micropython/micropython #2720: Can't get property value of the superclass

nupha, 2016-12-25:
As following code shows, a subclass to override property 'p' of superclass. Calling super().p in subclass will return a property object, not the value. Further more, this property object has no .fget attribute as suggested in standard python. But a 'getter' method is found on property object, when I call .getter(o), another property object returned, still not value. 

So, I can't get the property value of superclass. Any suggestion?

class A:
    @property
    def p(self):
        return {"a":10}

class AA(A):
    @property
    def p(self):
        # get super property, add some new value
        v = super().p
        v['aa'] = 20    # this will raise an exception, because v is a property object, not a dict
        return v

peterhinch, 2016-12-25:
This can be simplified to
```python
class A:
    @property
    def p(self):
        return {"a":10}

class AA(A):
    @property
    def p(self):
        return super().p
```
Under CPython 3.5.2:
```
>>> a = AA()
>>> a.p
{'a': 10}
>>> 
```
Under MicroPython (Unix build)
```
>>> a = AA()
>>> a.p
<property>
>>> 
```
I can't see a reference to this in the [differences from CPython doc](https://github.com/micropython/micropython/wiki/Differences).

dpgeorge, 2016-12-29:
The problem is that when super() looks up names in the base class(es), if it finds something then it doesn't do post-processing on the result to check for properties/descriptors.  There is code in mp_obj_instance_load_attr to do this but it would need to be factored out.

maruno, 2023-03-05:
Came across this issue and am wondering if there's any workaround. Could I ask the property-object that gets returned to execute the getter in some way?