micropython/micropython #2722: Christmas Issue - Filesystem read and write corrupts filesystem

cefn, 2016-12-25:
We recently used Micropython on NodeMCU v2 to create the Morecambe Minilluminations... 
See e.g. http://res.cloudinary.com/jpress/image/fetch/w_700,f_auto,ar_3:2,c_fill/http://www.thevisitor.co.uk/webimage/1.8275541.1481016719!/image/image.jpg or 
http://www.thevisitor.co.uk/news/lanterns-light-the-way-for-morecambe-festival-1-8275546

A friend wanted one for her street advent calendar...
https://twitter.com/ShrimpingIt/status/808995712809762816
...so we made an animated Xmas tree (this video doesn't give the colours justice)....
https://www.youtube.com/watch?v=UQFYyksohB4

She wasn't sure if she always wanted animated or not, so instead of requiring her to access the [embedded REPL web interface](https://github.com/ShrimpingIt/cockle/tree/master/replserver) for every time she wanted to switch, I wrote a fragment of code which read then wrote config values into the filesystem to provide simple JSON persistence... (see declarations and invocations of functions with Config in the name)...
https://github.com/ShrimpingIt/tableaux/blob/4e795a63e1d83ca07439c18e5789ba93091b785d/regimes/xmastree/main.py
...then used this persistence to create a boot counter. 

Odd values of the boot counter would trigger static behaviour, and even numbers would trigger animated behaviour, meaning she just had to turn it off and on again to switch between behaviours.

This ran for roughly 8 days through a few cycles of her turning it on and off. After 8 days she reported that the tree had stopped lighting, and I feared a fatal short circuit from possibly overvolting the red LEDs. However, it turned out on inspection that the REPL was still live and there were no electronic issues, but the config file loading request had failed because the filesystem was corrupted.

Any idea what I can do to mitigate this corruption? 

Have I misunderstood something fundamental about the filesystem? 

Are there workarounds like adding sleeps between boots/reads/writes. 

Is there a preferred way to implement the persistence needed for a boot counter?

peterhinch, 2016-12-26:
The code looks OK to me. I wonder if there are occasional electrical problems on switch-on: you are writing to the filesystem immediately after a boot. I'd be tempted to insert a delay of a few seconds to give the power time to stabilise. Questions like this are best raised in the forum, by the way: this is really for firmware bug reports.

cefn, 2016-12-26:
Thanks, @peterhinch good suggestion. I figured that filesystem corruption was bound to be a firmware issue of some kind (or an undocumented interaction between read and write which could be noted) but I'll investigate the potential of electrical instability. Worth wiring up  the rig with the mains USB power supply I'm using to the oscilloscope, for example. Figured the LDO regulator was keeping a stable 3.3V but if that could take a while to stabilise it would be an explanation. A boot delay workaround is prob worthwhile too. Starting with the forum is a good idea.

pfalcon, 2017-04-05:
Any updates on this? Unless there's an way to exactly reproduce this on different modules at will, this indeed would be classified as a hardware (specifically, power) instability issue (and a well-known, common one).

closed 2017-05-14 by pfalcon