micropython/micropython #2725: cc3200: UART REPL is broken

pfalcon, 2016-12-26:
As 1c7f9b16f0037cf3ca206de1f464d8237587ded7 says, UART REPL was deliberately "disabled" in cc3200 port. But it's actually not even disabled, it's not supported - grep for MICROPY_STDIO_UART finds it only in cc3200/boards/LAUNCHXL/mpconfigboard.h.

This of course makes initial start with the port, and development/testing unduly complicated. Support for serial REPL should be restored and it should be the default.

pfalcon, 2016-12-26:
Addressed in a quick manner in 5973098306fe1d9ed5b2db3cbc9e73370624505e.

dpgeorge, 2016-12-28:
> Addressed in a quick manner in 25f44c1.

IMO this is the correct thing to do.  On my WiPy 1.0 I have a boot.py script written long ago which has this os.dupterm line (along with wlan config settings).  My boot.py script has not changed in ages, nor been overwritten, hence why I never noticed this issue.

pfalcon, 2016-12-28:
Good. Indeed, a port should be in a provably working condition, and a way to prove that is to run unit testsuite, and it should be possible to do that over standard simple wire interfaces.

Enabled by default in 076b80467b74cbdfca5585d43e01a01ba4f8ff9c . Yet to do: update REDME.

pfalcon, 2016-12-28:
> Yet to do: update REDME.

Done in 514b82900c02808b8ad9bf5ecaa0e5bff270b33c. @dpgeorge , please have a look, because I just quickly looked at the WiPy schematics.

closed 2016-12-28 by pfalcon

pfalcon, 2016-12-29:
> Yes it looks correct.

It looks correct == there's no way to flash WiPy over serial? I.e. if you flash broken firmware, WiPy is bricked? Or are there really 3 firmware area, one of which is never OTAed and is booted on safe mode? Schematics have SOP2 "test point", is it visible/accessible on the actual board?

danicampora, 2016-12-29:
SOP2 on the WiPy is only accessible via the test points on the bottom 
side of the PCB, we used a bed of nails to flash them. On the other 
hand, the CC3200-Launchpad can be programmed over serial by just moving 
some jumpers.

There are 3 firmware areas, there's no way to brick it via an OTA update 
as the factory firmware is never overwritten and you can select it by 
safe booting.

pfalcon, 2016-12-29:
@danicampora : Ok, thanks for confirming.