micropython/micropython #2730: SD Card: ExFAT and Multipartition access

r4d10n, 2016-12-27:
FatFS v0.12(b) have added ExFAT support and we have a requirement to support >32 GB sdcards. I have been porting the code in the lib/fatfs to support the latest release of Fatfs.

I have almost merged all the changes made to the original library (/* dpgeorge: ... */ comments really helped!). But I am facing this spurious error due to bad optimization while linking, which have been reported elsewhere also.

> LINK build-STM32F7DISC/firmware.elf
> build-STM32F7DISC/lib/fatfs/ff.o: In function `f_lseek':
> ff.c:(.text.f_lseek+0x104): undefined reference to `__aeabi_uldivmod'
> ff.c:(.text.f_lseek+0x118): undefined reference to `__aeabi_uldivmod'
> Makefile:323: recipe for target 'build-STM32F7DISC/firmware.elf' failed

From this [post](http://lists.infradead.org/pipermail/linux-arm-kernel/2011-April/049729.html), I am almost convinced that it is from the following code section in ff.c, where the compiler seems to optimize the divide and modulo(& ~(x-1) here) operation into a __aeabi_uldivmod() function call. 

```
bcs = (DWORD)fs->csize * SS(fs);	/* Cluster size (byte) */

if (ifptr > 0 &&
	(ofs - 1) / bcs >= (ifptr - 1) / bcs) {	/* When seek to same or following cluster, */
	fp->fptr = (ifptr - 1)  & ~(FSIZE_t)(bcs - 1);	/* start from the current cluster */
	ofs -= fp->fptr;
	clst = fp->clust;
```
I tried compiling ff.c with -O0 - no luck. Also tried adding `asm("" : "+r"(bcs));` as reported in the earlier mentioned post, which also didn't seem to help. If I comment out `(ofs - 1) / bcs >= (ifptr - 1) / bcs` from the if condition check, the code seems to compile fine.  

Similar section also exist in the old FatFs code in master, yet no issues. Any suggestions for a fix ? 

Also on a related yet off-topic note: How do I mount a Multipartition SD card, so that I can access the other partition in the disk from the REPL (say as /sd2) ? Both partitions are accessible through the USB MSC.

dhylands, 2016-12-27:
This [document](http://infocenter.arm.com/help/topic/com.arm.doc.ihi0043d/IHI0043D_rtabi.pdf)  describes the various aeabi functions.  __aeabi_uldivmod is a 64-bit divide/modulo function. Normally it is provided by libgcc.a Since the STM32 is a 32-bit cpu and doesn't have any native 64-bit arithmetic, it doesn't matter what optimization level is used, the function is still required.

One solution is to add libgcc.a to the linker step (I have Makefile snippets to find the correct libgcc.a). Another solution is to refactor the code so that it doesn't need to do a 64-bit divide.

I believe that libgcc may be left out for a reason, but I can't recall why. So we probably need @dpgeorge  to comment.

dpgeorge, 2016-12-28:
I have been working on improving the PR #2140 which includes an updated version of fatfs.  It's getting close to being finished.

In the meantime, if you really need ExFAT support, then try using libgcc.a as @dhylands suggests (see line 79 of stmhal/Makefile).

> Also on a related yet off-topic note: How do I mount a Multipartition SD card, so that I can access the other partition in the disk from the REPL (say as /sd2) ? Both partitions are accessible through the USB MSC.

Multipartition is currently not supported by uPy.  You can try to get it to work by playing with the numbers in VolToPart in stmhal/fatfs_port.c.  Try changing the line `{1,0}` to `{1,1}` and the line `{2,0}` to `{1,2}`.

r4d10n, 2016-12-28:
Uncommenting line 79 of stmhal Makefile gave me this:

```
arm-none-eabi-ld: error: build-STM32F7DISC/firmware.elf uses VFP register arguments, /usr/bin/../lib/gcc/arm-none-eabi/6.2.1/libgcc.a(_udivmoddi4.o) does not
arm-none-eabi-ld: warning: /usr/bin/../lib/gcc/arm-none-eabi/6.2.1/libgcc.a(_udivmoddi4.o) uses variable-size enums yet the output is to use 32-bit enums; use of enum values across objects may fail
arm-none-eabi-ld: failed to merge target specific data of file /usr/bin/../lib/gcc/arm-none-eabi/6.2.1/libgcc.a(_udivmoddi4.o)
Makefile:325: recipe for target 'build-STM32F7DISC/firmware.elf' failed
```
`arm-none-eabi-gcc -print-libgcc-file-name` was pointing to `/usr/bin/../lib/gcc/arm-none-eabi/6.2.1/libgcc.a`. Manually pointing the Makefile to ` /usr/lib/gcc/arm-none-eabi/6.2.1/hard/libgcc.a`
did the job. 

I have working ExFAT now, with doubled write speeds (~8MBps). I'm testing a full 64GB write now. 

If there is any immediate interest, I could clean up a patch, or we can wait for PR #2140.

Still playing with Multipartition. As mentioned earlier, the USB MSC from uPy is able to detect the volumes and mount the two partitions in the host. So I guess I need to figure out how to f_mount the partition as /sd2.

dhylands, 2016-12-28:
Line 79 needs to be modified from:
```
LIBS += $(shell $(CC) -print-libgcc-file-name)
```
and add CFLAGS
```
LIBS += $(shell $(CC) $(CFLAGS) -print-libgcc-file-name)
```
This will result in a linker warning:
```
enums; use of enum values across objects may fail
arm-none-eabi-ld: warning: /home/dhylands/stm/gcc-arm-none-eabi-4_9-2015q3/bin/../lib/gcc/arm-none-eabi/4.9.3/armv7e-m/fpu/libgcc.a(_udivdi3.o) uses variable-size enums yet the output is to use 32-bit
```
This warning shouldn't be a problem as libgcc.a doesn't use enums. It is posisble to get rid of the warning by remove the `-mabi=aapcs-linux` option from CFLAGS and rebuild everything.

Note that removing -mabi=aapcs-linux causes the firmware size to increase by about 1K.

dpgeorge, 2016-12-29:
> Still playing with Multipartition. As mentioned earlier, the USB MSC from uPy is able to detect the volumes and mount the two partitions in the host. So I guess I need to figure out how to f_mount the partition as /sd2.

You can use the pyb.SDCard() class to read raw blocks from the SD card.  If you read the first block you can write some Python code to decode the partition table that lives there.  Then you can create your own block device class that exposes a section (ie partition) of the pyb.SDCard() class.  You can then mount this as usual with os.mount() and you'll get the second partition.

dpgeorge, 2017-01-27:
The FatFs library is now at version R0.12b (see lib/oofatfs/).  But it's not yet usable with the stmhal port.

dpgeorge, 2017-01-27:
Since 84c614e7290804e8b8d7ebff03ca35a4a4c5fcf1 stmhal uses FatFs code based on R0.12b, so should support exFAT (but it's untested).

dpgeorge, 2017-01-31:
Multiple partitions are now supported in stmhal: when mounting the SD card on boot-up all available FatFs partitions are mounted as /sd, /sd2, /sd3, /sd4.  See 3667ee1b8870ddc76f47260eb87d7d13736c9ad2

There is currently no support for the user mounting different partitions themselves.  That would require implementing a function like os.partition_device(bdev) which returns a tuple of block devices, one for each partition in the given bdev.  Something for the future.

Regarding exFAT: I've tested it on stmhal and it works.  It's disabled by default because the technology is subject to licensing from Microsoft.  You can enabled it for your own builds/devices by defining MICROPY_FATFS_EXFAT to 1 in mpconfigboard.h.

dpgeorge, 2017-01-31:
> I believe that libgcc may be left out for a reason, but I can't recall why. So we probably need @dpgeorge to comment.

It was because, at the very start of the project, I didn't want to use any external functions/libraries, so I know exactly what was going on and what functions were necessary to run on baremetal.  But libgcc adds useful low-level stuff like uldivmod so I guess it would be good to include it now that we are finding a need for it (eg we wouldn't need to patch the HAL SD card driver to remove long division).

> This warning shouldn't be a problem as libgcc.a doesn't use enums. It is posisble to get rid of the warning by remove the -mabi=aapcs-linux option from CFLAGS and rebuild everything.
> 
> Note that removing -mabi=aapcs-linux causes the firmware size to increase by about 1K.

The reason I went with aapcs-linux is because I wanted a definite, known ABI so that the hand-written assembler code would work (eg NLR, GC register scanning, native code generation).  But I see that aapcs has the same ABI (EABI) as aapcs-linux, and the only difference is that aapcs uses variable enums.  According to http://kanj.github.io/elfs/book/armMusl/cross-tools/abi.html (sec 6.4.2) aapcs-linux is recommended, but it doesn't say why.

I'd be happy to change to aapcs so that we can link against the standard libgcc.  Or better still, just remove the -mabi definition altogether and let the compiler choose the "standard" one according to it.

I do see an increase of about 1k in flash using aapcs, but also a *decrease* in the BSS of about 60 bytes.

dpgeorge, 2017-02-08:
aapcs-linux was dropped, and libgcc linked by default, in 7ae9bee7905601a827f3d216300f917afbacf07e

dpgeorge, 2017-06-14:
Main issues raised by first comment are now implemented (optional exFAT support and multipartition SD mounted on boot up), so closing.

closed 2017-06-14 by dpgeorge