micropython/micropython #2734: UnicodeDecodeError not raised when expected in bytes.decode()

hiway, 2016-12-28:
**What is expected:**

Python 3.5.2/ 3.6.0:

    >>> bytes.decode(b"\xa1\x80", 'utf-8')
    Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
    UnicodeDecodeError: 'utf-8' codec can't decode byte 0xa1 in position 0: invalid start byte

**What is happening:**

MicroPython v1.8.6-260-gafc5063-dirty on 2016-12-28; darwin version:

    >>> bytes.decode(b"\xa1\x80", 'utf-8')
    '\u0840'

I am porting umsgpack (a small pure-python msgpack library, the 'u' is not related to upy) to micropython, and [this particular test is failing](https://github.com/hiway/u-msgpack-python/pull/1/commits/fdc6951a0b7d779e0bc7304fb96501582aa55af9) since Micropython behaves differently from CPython. It may show up elsewhere as surprises if programs continue when they should fail.

pfalcon, 2016-12-29:
It's probably worth (for interested parties) to prepare the simplest possible patch to catch this issue, and then eyeball it for how much it takes.

chuckbook, 2017-01-02:
I'm using this code snippet...
```
/*
 * check for valid utf-8
 */
int utf_8_chk(const uint8_t *p) {
    uint8_t ch;
    uint8_t need = 0;
    while ((ch = *p++)) {
        if (need) {
            if (ch >= 0x80) {
                need -= 1;
            } else {
                // missmatch
                return 0;
            }
        } else {
            if (ch >= 0xc0) {
                if (ch >= 0xf8) {
                    return 0;
                }
                need = (0xe5 >> ((ch >> 3) & 0x6)) & 3;
            } else if (ch >= 0x80) {
                // missmatch
                return 0;
            }
        }
    }
    return (need == 0);     // no pending fragments allowed

dpgeorge, 2017-09-06:
This is now implemented by 68c28174d0e0ec3f6b1461aea3a0b6a1b84610bb.  Thanks @chuckbook for the function to check utf8.

closed 2017-09-06 by dpgeorge