micropython/micropython #2735: Prepare for release v1.8.7

dpgeorge, 2016-12-28:
The December release was skipped, but there have been enough changes to warrant a new release ASAP and it would be good to do it in the first week of January 2017.  The biggest change was the update of the HAL for stmhal, and bugs introduced by that have now been fixed (all the ones I know of).

Since there were no big user-facing changes going up a sub-minor version to v1.8.7 makes sense.

pfalcon, 2016-12-29:
Sounds good. I can think of 2 things worthy to fit in the release:

* Make cc3200 deployable with cc3200tool https://github.com/micropython/micropython/issues/2627
* Perhaps, make unix uselect changes as discussed in https://github.com/micropython/micropython-lib/pull/141#issuecomment-269595603, as it's getting harder and harder to sync micropython and micropython-lib package releases.

dpgeorge, 2016-12-30:
> Perhaps, make unix uselect changes as discussed in

It would be ideal to use extmod/moduselect.c for the unix port, but that would come at the expense of efficiency because the poll/select call becomes essentially a busy-wait loop.

Also, #2672 (esp native code location) would be good to have done.

pfalcon, 2016-12-30:
> It would be ideal to use extmod/moduselect.c for the unix port, but that would come at the expense of efficiency because the poll/select call becomes essentially a busy-wait loop.

I'm writing a long update to https://github.com/micropython/micropython/issues/1550, so long, that I couldn't even finish it in one night, and my current idea (still vague) is that vice-versa, it would be better for extmod/moduselect.c to adopt a design of unix', i.e. use an adhoc table to hold poll descriptors instead of mp_map_t. Anyway, I'll post it in some time, and we can consider what's the next logical step.

> Also, #2672 (esp native code location) would be good to have done.

Well, all infra was done for that. Feel free to modify flashbdev.py to include a setting to offset start of FS (I suggest using a generic name like RESERVED_SECS), then just add a docs para that native code cache area starts at esp.flash_user_start() and of size 4K. (Maybe we should rename esp.flash_user_start() to something shorter, I had esp.fw_end() in mind).

dpgeorge, 2017-01-04:
> Feel free to modify flashbdev.py to include a setting to offset start of FS

Done in bae7798f1ef531c862421b37c0d5d462e34a6d2f.

> add a docs para that native code cache area starts at esp.flash_user_start() and of size 4K

The esp.set_native_code_location() func is documented by c3f70c603efcb053810dbee8d5cc9f23ed62d01f.

> Maybe we should rename esp.flash_user_start() to something shorter, I had esp.fw_end() in mind

I kinda got used to flash_user_start.  It's self documenting in that it tells you you can start using the flash from this offset.

pfalcon, 2017-01-05:
Is there specific ETA for the release? It would be nice to get https://github.com/micropython/micropython/pull/2747 in and fix improperly working hwapi examples.

dpgeorge, 2017-01-05:
@pfalcon there's #2743, and also what do you think of https://github.com/micropython/micropython/commit/bae7798f1ef531c862421b37c0d5d462e34a6d2f#commitcomment-20367399

dpgeorge, 2017-01-05:
> wonder how much more beyond that would fit.

Not much.  I have the release almost ready to go, just want to make sure flashbdev changes to esp8266 are ok and then release should happen tomorrow.

pfalcon, 2017-01-06:
> For the release I suggest moving all symlinks from scripts/ to modules/ to make enough room for the firmware. Regarding RESERVED_SECS, it would be good not to require users to re-build their filesystem, so maybe revert this change (just do RESERVED_SECS=0) for this release...

Well, I guess it's good enough compromise, due to echo effect, it will take some time before people will start to play with native code...

dpgeorge, 2017-01-06:
Ok, I did as said above: change RESERVED_SECS to 0, updated docs, and moved all script symlinks to modules/ for esp8266 port (remaining flash space in esp-extra-scripts branch build is about 3k).  esp-extra-scripts branch has been rebased and force pushed.

Release notes have been written.  It just remains to do a bit of testing on esp8266 to make sure the last-minute changes didn't break anything.

mzdaniel, 2017-01-07:
MicroPython v1.8.6-302-gef1bbad-dirty on 2017-01-07; ESP module with ESP8266

Compilation went well (just a few warnings here and there). In the gist https://gist.github.com/mzdaniel/f7784768054300b2ab885e2ed32c9fda are the results of running the tests in micropython unix and esp8266. For testing in esp8266, I run in the repl micropython-lib/uasyncio/example_http_server.py (modifying the server socket to 0.0.0.0:8081). With that config, esp8266 pull 1000 queries with no concurrency, and 10 queries with 2 concurrent requests.

Will be nice to streamline testing in the esp8266. Happy to help.

pfalcon, 2017-01-07:
@mzdaniel : Standard way to test micropython (any port of) is using tests/run-tests script. See http://forum.micropython.org/viewtopic.php?f=16&t=1731 for more info.

mzdaniel, 2017-01-07:
Fantastic!  Added esp8266 run-tests to the gist. 473 passed, 35 skipped, 3 failed: viper_binop_arith viper_binop_multi_comp viper_misc.

pfalcon, 2017-01-07:
Ack, I get the same results, with 3 viper tests failing. @dpgeorge : I'm not sure if you plan to fix those, but I'd say as that's new and niche feature, ok to make a release with those failures.

dpgeorge, 2017-01-08:
> I'm not sure if you plan to fix those, but I'd say as that's new and niche feature, ok to make a release with those failures.

They fail only because they run out of iram1 memory to store the native code.  Using flash to store native code, and these viper tests all pass (at least they did the last time I checked).  So there's not much to do to fix them right now.

Otherwise, testing some more things on esp8266 I found an issue with import of frozen packages, now fixed in b528e9a428f92241d2cf3bb30f78b0d65e1a8428.

BTW, @mzdaniel , thanks for testing!

dpgeorge, 2017-01-08:
v1.8.7 was released and tagged in 5653e3c72fc8555c6a060acf6447ac694a036053

closed 2017-01-08 by dpgeorge

mkarliner, 2017-01-08:
Congratulations!