micropython/micropython #2742: making mpy-cross

BLCKPSTV, 2016-12-29:
Making mpy-cross gives me make: *** mpy-cross: No such file or directory.  Stop.

The dependencies where installed correctly without problems. 
Wemos D1 Mini pro 16mb

dpgeorge, 2016-12-29:
You need to build mpy-cross from within the mpy-cross directory of this repository.  So you either do `cd mpy-cross` and then `make`, or just `make -C mpy-cross` from the root of this repository.

dhylands, 2017-01-01:
```
make -C ../../../mpy-cross target
```
(use appropriate number of ..'s)

BLCKPSTV, 2017-01-01:
So I'm making the micro-python on for the ESP via this [PFalcon](https://github.com/pfalcon/esp-open-sdk) and to flash it on a ESP8266 device, it requires me to build a case-sensitive image. I manage to clone and build the [repository](git clone --recursive https://github.com/pfalcon/esp-open-sdk.git). And even managed to make/build it. 

From that moment I could go back to the build instruction in the subfolder of the ESP8266 folder in this github [here](https://github.com/micropython/micropython/tree/master/esp8266)

git submodule update --init worked. But then the `make -C mpy-cross` doesn't work. So I downloaded the master folder and tried to make the the mpy-cross from that folder. But my thought was, that it has to be in the case sensitive folder. 
But even the `make -C mpy-cross` didn't got me to build it. I got this : `Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.`

I was able to make it so far with little knowledge of python en and the command line. Really stuck at it. 

The `make -C ../../../mpy-cross target` didn't work as well because there is no top laying folder available in the image.

dpgeorge, 2017-01-03:
It sounds like you have an unconventional build environment, and it's likely not a bug with the code but with your set up.  Please ask for help at forum.micropython.org and specify your complete build environment and steps.

closed 2017-01-17 by dpgeorge