micropython/micropython #2744: stmhal pyb.Timer callback fails if it is a method

peterhinch, 2016-12-30:
Is it not OK for callbacks to be methods? It works with Pin objects but not with a Timer:
```python
import micropython
from pyb import Timer
micropython.alloc_emergency_exception_buf(100)

class Foo():
    def __init__(self):
        self.timer = Timer(4)
        self.timer.callback(self.cb_timer)
        self.timer.init(freq = 10)
        self.bar = True

    def cb_timer(self, tim):
        print('OK')
        if self.bar: # falls over
            print('.')

Foo()
```
Outcome:
```
MicroPython v1.8.1-1055-gafc5063-dirty on 2016-12-28; PYBv1.1 with STM32F405RG
Type "help()" for more information.
[paste omitted]
OK
uncaught exception in Timer(4) interrupt handler
Traceback (most recent call last):
  File "<stdin>", line 14, in cb_timer
AttributeError: 'Foo' object has no attribute 'bar'
<Foo object at 20003500>
>>> 
```

dhylands, 2016-12-30:
This works in general. I'm going to guess that your callback wound up getting called before self.bar=True was done in the constructor.

You basically have to assume that the callback can be called anytime after init is called. Here's an example I have:
https://github.com/dhylands/upy-examples/blob/master/heartbeat_irq.py

peterhinch, 2016-12-30:
You're right - undoubtedly I should have initialised bar first. I'm surprised, though, that with a period of 100ms this occurred. I'd have thought the first callback would occur 100ms after issuing init(). The failure implies that the callback occurred almost immediately. Is that expected behaviour?

dhylands, 2016-12-30:
Technically, the callback can occur anytime after callback was called. So it might even be happening before init.

If the timer was previously running, then there may have been a timer interrupt pending. You can also pass the callback= to the init function, that way the callback won't happen until after the timer is initialized.

peterhinch, 2016-12-30:
OK, thanks.

closed 2016-12-30 by peterhinch