micropython/micropython #2745: What version of eep-open-sdk is needed?

nameofuser1, 2016-12-31:
I have tried three versions 1.3, 1.5.4 and 2.0 and no one of them contains liblwip_open.a only liblwip.a or liblwip_536.a
So what version should I use to obtain necessary library?

Thanks.

robert-hh, 2017-01-01:
Did you follow all steps of of the build instructions at https://github.com/micropython/micropython/blob/master/esp8266/README.md, especially the step:
`git submodule update --init
`
after cloning the eep-sdk?

nameofuser1, 2017-01-01:
@robert-hh My fault I have found it in another directory. Just needed to move it in the right place. 

In order not to create new issues, are you able to compile it with -Werror flag? I can't because of the error:
 `hspi.c:96:9: note: in expansion of macro 'PIN_FUNC_SELECT'
         PIN_FUNC_SELECT(PERIPHS_IO_MUX_MTDI_U, 2);
         ^
/Volumes/case-sensitive/esp-open-sdk/sdk/include/eagle_soc.h:273:38: warning: suggest parentheses around arithmetic in operand of '|' [-Wparentheses]
                                      &  (~(PERIPHS_IO_MUX_FUNC<<PERIPHS_IO_MUX_FUNC_S))  \
                                      ^
/Volumes/case-sensitive/esp-open-sdk/sdk/include/eagle_soc.h:69:98: note: in definition of macro 'WRITE_PERI_REG'
 #define WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)ETS_UNCACHED_ADDR(addr))) = (uint32_t)(val)`

I fixed it but there is nothing about it in the guide.

robert-hh, 2017-01-01:
If you follow the install instructions cite above, you do not have to move or fix anything. About your first question: sdk 2.0 is needed for version 1.8.6 of MP (see http://forum.micropython.org/viewtopic.php?f=16&t=2140)

closed 2017-01-01 by nameofuser1