micropython/micropython #2746: ESP8266 Micropython Crash

straga, 2017-01-01:
MicroPython v1.8.6-7-gefd0927 on 2016-11-10; ESP module with ESP8266
 
file: irq.py

```
from machine import Timer

frequency = 0

def fget():
    global frequency
    return frequency

def addcount(p):
    global frequency
    frequency += 1

def init():
    timer1 = Timer(-1)
    timer1.init(period=1000, mode=Timer.PERIODIC, callback=addcount)
```

console:

import irq
irq.init()

Call irq.fget() many time by hand, and esp reboot may be after 3 or 5 call.
[Video](https://youtu.be/UUaOSiOcipI)

robert-hh, 2017-01-01:
I had a variation of the outcome, in that the counter got stuck at a certain value:
```
>>> import irq
>>> irq.init()
>>> irq.fget()
5
>>> irq.fget()
8
>>> irq.fget()
16
>>> irq.fget()
17
>>> irq.fget()
17
>>> irq.fget()
17
>>> irq.fget()
17
>>> 
```
MicroPython v1.8.6-261-g076b804-dirty on 2016-12-28; ESP module with ESP8266
On other attempts, the esp8266 falls into reset:
```
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)
wdt reset
```
In any case, not the expected behavior.
B.t.w: Happy new year for all of you.

dpgeorge, 2017-03-20:
I couldn't reproduce this with the latest code, but it should anyway be fully fixed by the new "soft irq" handler; see #2878.

closed 2017-03-20 by dpgeorge