micropython/micropython #2749: dict update fails when called with dict subclass

stinos, 2017-01-02:
Given

```
class MyDict(dict):
  pass

print(MyDict(MyDict(a=1, b=2)))
```

it's expected (well, per CPython) that the 'outer' MyDict is populated with the key/value pairs from the 'inner' dict. This doesn't work in uPy as it uses ```MP_OBJ_IS_DICT_TYPE``` internally to check is MyDict is a dict and if not resorts to treating the inner instance as an iterable of pairs:
```
Traceback (most recent call last):
  File "foo.py", line 4, in <module>
ValueError: dictionary update sequence has the wrong length
```

dpgeorge, 2017-01-03:
CPython looks like it's treating subclassing of dict here as a special case.  There doesn't seem to be any way to create a custom class with a set of special methods that allows it to act like a dict (a Mapping) and initialise a dict as above.  The only way to create such a custom class is to derive from dict.  So uPy would need a similar specialised check to make it behave the same.

stinos, 2017-01-03:
I checked CPython and what they do is first check if the object passed in has a `keys` attribute. Then if it's a dict subclass a fast update path is used, else a slower generic one which is basically
```
for k in b.keys():
  a[s] = b[x]
```
That seems a bit much for uPy, but implementing a subset of that functionality (i.e. make update work if passed a dict subclass) doesn't seem too hard:
```
@@ -363,13 +363,18 @@ STATIC mp_obj_t dict_update(size_t n_args, const mp_obj_t *args, mp_map_t *kwarg

     if (n_args == 2) {
         // given a positional argument
-
+        mp_obj_dict_t *rhs;
         if (MP_OBJ_IS_DICT_TYPE(args[1])) {
+            rhs = (mp_obj_dict_t*)MP_OBJ_TO_PTR(args[1]);
+        } else if(mp_obj_is_subclass_fast(MP_OBJ_FROM_PTR(mp_obj_get_type(args[1])), MP_OBJ_FROM_PTR(&mp_type_dict))) {
+            rhs = ((mp_obj_instance_t*)MP_OBJ_TO_PTR(args[1]))->subobj[0];
+        }
+        if (rhs) {
             // update from other dictionary (make sure other is not self)
             if (args[1] != args[0]) {
                 mp_uint_t cur = 0;
                 mp_map_elem_t *elem = NULL;
-                while ((elem = dict_iter_next((mp_obj_dict_t*)MP_OBJ_TO_PTR(args[1]), &cur)) != NULL) {
+                while ((elem = dict_iter_next(rhs, &cur)) != NULL) {
                     mp_map_lookup(&self->map, elem->key, MP_MAP_LOOKUP_ADD_IF_NOT_FOUND)->value = elem->value;
                 }
             }
```
I'm not sure this is the correct way to access subobj, and more importantly: I'm not sure if it's useful for anyone else? It only solves my particular problem but I can deal with it in Python code as well.

MrSurly, 2017-06-01:
Okay, so I've hit this issue as well.  My use case is a "smart" config object that knows how to read/write a JSON file (among other things).  

I ran into this when I was working on the "save" code -- I didn't think that passing the object to the `json.dumps()` method would work, so I wrapped it: `dict(self)`.

Turns out that passing `self` directly into `json.dumps()` works just fine.

```python
    def save_config(self):
        try:
            with open(self.config_filename, 'w') as f:
                f.write(json.dumps(dict(self)))
        except Exception as e:
            print('Unable to save config', e)
```

projectgus, 2024-08-28:
Is this best resolved by documenting the difference to CPython?