micropython/micropython #2755: TypeError: argument should be a 'object' not a 'DER'

dmazzella, 2017-01-03:
I'm working on implementing object .__ setattr__ and object .__ delattr__ with this modification:
[objobject.patch](https://github.com/micropython/micropython/files/682661/objobject.txt)

and a new macro `#define MICROPY_PY_METHODS_DELATTRS_SETATTRS (1)` into unix/mpconfigport.h

using this code:

```
class DER(object):

    def __init__(self, typeid=0, classid=0, indefinite=0, value=b''):
        object.__setattr__(self, "_value", value)

der = DER()
der.c = 2
print(der.c)
```

i have this results:

```
dma@dma-pc MSYS ~/micropython/unix
$ ./micropython ../tests/basics/object__setattr__.py
Traceback (most recent call last):
  File "../tests/basics/object__setattr__.py", line 12, in <module>
  File "../tests/basics/object__setattr__.py", line 6, in __init__
TypeError: argument should be a 'object' not a 'DER'

dma@dma-pc MSYS ~/micropython/unix
$ python3 -B ../tests/basics/object__setattr__.py
2
```

the problem is into [runtime.c](https://github.com/micropython/micropython/blob/master/py/runtime.c#L935
), the solution can be add this check:

`if (arg0_type != self->type && !MP_OBJ_IS_TYPE(arg0_type, &mp_type_type)) {`

is the correct solution for this?

pfalcon, 2017-01-03:
Which leads to a question - what's the purpose of all these changes? https://github.com/micropython/micropython/pull/2753, this, and more, because it won't stop here of course. Is it to make MicroPython better, or to run that funkily implemented "DER" class?

At the same time, this situation was well described couple of years ago in https://github.com/micropython/micropython/wiki/ContributorGuidelines#django-criteria . Let me quote it for you:

> If you port existing code, you'll likely find out that a lot of features missing, so adding all of them will be a lot of work, and will bloat MicroPython, and adding just one of few won't make a big difference. (So, just keep porting, by patching existing code to use facilities MicroPython already provides.)

@dmazzella , can you please explain why instead of following that advice, you choose to patch MicroPython?

dmazzella, 2017-01-03:
the purpose of all these changes is split #2689

dmazzella, 2017-01-06:
if i add this check to [runtime.c](https://github.com/micropython/micropython/blob/master/py/runtime.c#L935):

```
if (arg0_type != self->type
            #if MICROPY_PY_OBJECT_METHODS_DELATTRS_SETATTRS
            && !mp_obj_is_subclass_fast(MP_OBJ_FROM_PTR(arg0_type), MP_OBJ_FROM_PTR(&mp_type_object))
            #endif
            ) { ...
```

but not directly subclass "object":

```
>>> class A:
...     pass
...
...
...
>>> a = A()
>>> object.__setattr__(a, "a", 1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: argument should be a 'object' not a 'A'
>>>
```

with subclass:

```
>>> class A(object):
...     pass
...
...
...
>>> a = A()
>>> object.__setattr__(a, "a", 1)
1
>>>
```

Any reason for this?

dmazzella, 2017-01-06:
if i add this check to [objtype.c](https://github.com/micropython/micropython/blob/master/py/objtype.c#L901):

```
    #if MICROPY_PY_OBJECT_METHODS_DELATTRS_SETATTRS
    if (len == 0) {
        // if it does not inherit from any type, replace 'bases_tuple' 
        // with a tuple containing one element of type 'object'
        STATIC const mp_rom_obj_tuple_t type_base_tuple = {{&mp_type_tuple}, 1, {MP_ROM_PTR(&mp_type_object)}};
        bases_tuple = MP_OBJ_FROM_PTR(&type_base_tuple);
    }
    #endif
```

and add this to [runtime.c](https://github.com/micropython/micropython/blob/master/py/runtime.c#L935
):

```
    if (arg0_type != self->type
        #if MICROPY_PY_OBJECT_METHODS_DELATTRS_SETATTRS
        && !mp_obj_is_subclass_fast(MP_OBJ_FROM_PTR(arg0_type), MP_OBJ_FROM_PTR(&mp_type_object))
        #endif
        ) { ...
```

the result are:

for cPy

```
>>> class A:
...     pass
...
>>> a = A()
>>> object.__setattr__(a, "a", 1)
>>> a.a
1
>>>
```

for uPy

```
>>> class A:
...     pass
...
...
...
>>> a = A()
>>> object.__setattr__(a, "a", 1)
>>> a.a
1
>>>
```

This solution may be acceptable?

dmazzella, 2017-01-10:
@dpgeorge any hints?

dpgeorge, 2017-01-11:
> any hints?

No, not at this time.  This stuff is low priority, sorry.

pfalcon, 2017-05-14:
Should this ticket still be open? "DER" type the title refers to is not part of MicroPython. Please update the title to describe the real problem if this ticket should be open.

dpgeorge, 2017-05-15:
`__delattr__` and `__setattr__` were implemented for user defined classes in 18e65691661ef8e83060d0e72d66b16cc918c8b4.  The first script posted in this issue with the "DER" class now raises the error `AttributeError: type object 'object' has no attribute '__setattr__'` which shows the actual issue, that the base-class object doesn't support `__setattr__` (or the other related ones).

I don't know what the original intent of @dmazzella is with needing support for `object.__setattr__` but it may be that it can be achieved simply with the built-in `setattr`, like `setattr(self, "_value", value)`

dmazzella, 2017-05-15:
> I don't know what the original intent of @dmazzella is with needing support for object.__setattr__ but it may be that it can be achieved simply with the built-in setattr, like setattr(self, "_value", value)

My initial need was support object.__setattr__

dpgeorge, 2017-05-16:
> My initial need was support object.setattr

Ok, but is that supporting it just so it's there, or was there some code you tried that needed it, or did you write some code expecting it to be there?

pfalcon, 2017-05-16:
Well, the original title tells the story: @dmazzella took highly bloated, unportable to MicroPython, CPython code and decided to port it. It didn't work, but instead of the usual hint to patch the Python code he started to patch MicroPython. Let me tell that again: there's no, absolutely no, need to use `__setattr__` to implement DER encoding https://en.wikipedia.org/wiki/X.690 . If some code used it for that, it's a problem of that code and not of MicroPython.

dmazzella, 2017-05-16:
Ok, il no needs about object.__XXXattr__ i close it.

closed 2017-05-16 by dmazzella

NuclearManD, 2019-08-03:
Not sure if this is useful to anyone other than me, but I found a workaround and I thought it might as well go here since I never found the answer I needed here (or anywhere else).

A simple example to illustrate the premise:
```
>>> class test_class:
...     def __getattr__(self, n):
...         return "No, I'm here :wink:"
...         
...         
... 
>>> tst = test_class()
>>> tst.some_nonexistent_attribute
"No, I'm here :wink:"
```

I found that creating a __getattr__() function actually allows this to be done if one is clever.  I plan on using a dictionary in the object to store custom attributes that are created in __init__.  They cannot be predicted before that point, unfortunately.  I also found that setattr() does not work unless the attribute exists, so I needed this workaround.

cdwilson, 2019-10-03:
FYI, for those coming across this in the future, this forum discussion has some ideas for working around missing `object.__setattr__` https://forum.micropython.org/viewtopic.php?f=2&t=7028

If other folks have worked around this or there are examples of better patterns to use, feel free to add to the forum discussion (I'd be interested in learning about them).