micropython/micropython #2762: os.utime is missing

zsquareplusc, 2017-01-05:
When writing file sync tools, it would be nice if the files put on a micropython board could be updated with their original timestamp (via [os.utime](https://docs.python.org/3/library/os.html#os.utime)). So sync tools could check the file date to determine if a file needs to be changed or is already up-to-date. (A more precise check can be made with actually inspecting the file contents, but for simple and quick syncs, the date is preferable).

Currently it is only possible to create new files that will get the "current time" which is a more or less random value near 1970 or 1985, as the clocks of those boards are usually not set and not battery backed up.

I see the FAT driver already has a `f_utime()` function, so adding this one should be easy (I might provide a pull request with a basic implementation e.g. times in seconds not ns, if desired).

dhylands, 2017-01-05:
What I do with rshell, is that I have it set the RTC clock on the pyboard to match the time on the host computer. Then the timestamps on the filesystem will match the host (until the pyboard is powered off).

Of course, connecting a backup battery will allow the RTC to maintain time even if the pyboard is powered off.

From a file sync perspective, at boot time you could also just set the RTC to the time of the newest file on the filesystem (perhaps plus a few seconds), then at least the file would get updated with a "newer" (albeit incorrect) timestamp.

zsquareplusc, 2017-01-06:
That makes it easier for the user to see that the files are current but it gives you the time of copy, not the original file time, which I would prefer.

pfalcon, 2017-01-06:
Based on @dhylands' suggestion , the workaround should be obvious - set RTC time to the needed time every time before copying a file.

robert-hh, 2017-01-06:
In my main.py I have the following statements to set the RTC at boot time:
```
# get time
from ntptime import settime
settime()
```
The, files will have a valid time stamp.

rolandvs, 2017-01-06:
Nice to see that all versions behave differently

Inspired by @dhylands comments, I have a battery connected to the appropriate pin…
Need to be extended as is documented in the comments.

#boot.py

import pyb, time, os

# VIRTUAL COM PORT + MASS STORAGE
pyb.usb_mode('CDC+MSC')

# there is a real time clock that is active as long as the battery is connected and working
# At boot time we check if the time has been set (>2015) and if so continue with the program
# by displaying the time. If not set, then the existence of a file 'timesync' is checked. If
# so the creation time is read and the RTC is set accordingly.
# use scramble() to set the time to 2015, forcing setting the time again (for test)

rtc=pyb.RTC()
printformat = "{:02d}-{:02d}-{:4d} {:02d}:{:02d}:{:02d}"

def scramble():
    ''' scramble the rtc time '''
    rtc.datetime((2015,1,1,0,0,0,0,0))
    print(rtc.datetime())

def currenttime(leadingtext="Time is "):
    ''' print the time '''
    yr,mn,dy,wd,hr,mi,se,_ = rtc.datetime()
    print(leadingtext + printformat.format(dy,mn,yr,hr,mi,se) )

yr,_,_,_,_,_,_,_ = rtc.datetime() # ignore the rest
if yr==2015:
    # need to set the clock
    # could also check if the time of the file is newer than existing time (always)
    try:
        # an error will skip the rest
        _,_,_,_,_,_,_,_,_,emit = os.stat("timesync")
        newtime = time.localtime(emit)
        # time.localtime() is different from rtc.datetime()
        # (yr,mm,dy,hr,mi,se,wd,yd) versus (yr,mm,dy,wd,hr,mi,se,0)
        pyb_rtc_time = newtime[:3] + newtime[6:7] + newtime[3:6] + (0,)
        rtc.datetime(pyb_rtc_time)
        currenttime("Time is set to ")
    except:
        print("Time is not set")
        pass
else:
    currenttime("Today is ")

pyb.main('main.py')

pfalcon, 2017-05-06:
If there's a continued interest, this should be mentioned in https://github.com/micropython/micropython/issues/2920 for consideration of being an optional feature of VFS interface.

jonnor, 2024-09-08:
Does the VFS layer now set file creation based on RTC values? If so, I believe we can close this issue.

PS: in addition to setting RTC values using ntptime for networked devices, one can also set it from a host using the `mpremote rtc` command