micropython/micropython #2764: @staticmethod in native class

abiliojr, 2017-01-06:
I’m trying to natively implement (in C) something equivalent to this:

```
class MyClass(object):
    @staticmethod
    def the_static_method(x):
        print(x)
```

This class will be inside a module (the module has several classes). If I put the function straight in the module, the code works.

But if instead I try to put it inside the class, it will compile with no issues, but will give a: `TypeError: argument has wrong type`. Mind that the function expects an int:

```
STATIC mp_obj_t the_static_method(mp_obj_t x) {
    uint32_t n = mp_obj_get_int(x);

    printf(“%d\n”, x);
    return mp_const_none;
}
STATIC MP_DEFINE_CONST_FUN_OBJ_1(the_static_method_obj, the_static_method);
```

And the class dictionary is defined like:

```
STATIC const mp_map_elem_t MyClass_table[] = {
    { MP_OBJ_NEW_QSTR(MP_QSTR___name__),            MP_OBJ_NEW_QSTR(MP_QSTR_myClass)},
    { MP_OBJ_NEW_QSTR(MP_QSTR_delay_us),            (mp_obj_t)&delay_us_obj },
};

STATIC MP_DEFINE_CONST_DICT(MyClass, MyClass_table);
```

Is there any way to do the equivalent to `@staticmethod` while implementing the class in C?

pfalcon, 2017-01-06:
Grep codebase for "STATICMETHOD".

abiliojr, 2017-01-06:
Thanks! , after following your hint, made it work. Something like this worked:

 ```
STATIC mp_obj_t the_static_method(mp_obj_t x) {
    uint32_t n = mp_obj_get_int(x);

    printf(“%d\n”, x);
    return mp_const_none;
}
STATIC MP_DEFINE_CONST_FUN_OBJ_1(the_static_method_fun_obj, the_static_method);
STATIC MP_DEFINE_CONST_STATICMETHOD_OBJ(the_static_method_obj, &the_static_method_fun_obj);
```

closed 2017-01-06 by abiliojr