micropython/micropython #2765: OneWire - dallas sensor return positive value on minus degree

syndycat, 2017-01-06:
Hello,

I checked the onewire python code with DS18B20 - outside temperature -12 degrees celsius ... value returned by lib is 4098.

However inside was 23.5 degrees celsius and the value was OK

dpgeorge, 2017-01-06:
Which board/hardware/port are you using and where did you get the ds18b20 driver from?

peterhinch, 2017-01-07:
A search on DS18B20 in the forum yields posts on this issue and solutions.

syndycat, 2017-01-07:
Hello guys,

Myboard is pycom WiPy... but I beleive this doesn't matter . The digital msb / lsb is the same.

I fixed here https://github.com/micropython/micropython/pull/2770

Kind regards,
Cata

dpgeorge, 2017-01-08:
> Myboard is pycom WiPy... but I beleive this doesn't matter

The reason I asked is because there're different drivers for esp8266 and "all other boards", and the esp8266 driver had this bug fixed recently.

Hopefully soon we can merge the drivers so there is only one, and it works on all boards.  Until then I merged your fix, thanks!

closed 2017-01-08 by dpgeorge