micropython/micropython #2766: pfalcon, please unblock me...

kfricke, 2017-01-07:
...it is hindering my perception of this repository.  If you feel personally offended by me, please ignore me without restricting me this much. Thanks!

stinos, 2017-01-07:
Blocked from what, if I may ask?

kfricke, 2017-01-07:
Github notifications and all he does. As him being one of the maintainers, it sucks to only get noticed about only half of the great efforts which happen these days.

stinos, 2017-01-07:
I don't get it, I can select to watch/unwatch this repository and that's it. Are you saying a collaborator to a repository can somehow intervene with that? How? I don't even see a way of doing such thing for repositories I own.

kfricke, 2017-01-07:
Indeed. If you choose to block me (go to my profile and follow the link below the follow button on the left), than i can not follow/subscribe/star you or your repositories anymore. I will not receive anything you do on  Github.
https://help.github.com/articles/blocking-a-user-from-your-personal-account/

That of course is an important feature, but with people as maintainers of such an repository like MicroPython it is annoying for me.

kfricke, 2017-01-07:
Could one please @mention him here, thanks!

drawkula, 2017-01-08:
> Could one please @mention him here, thanks!

Ok... but not in front of a mirror: "@pfalcon, @pfalcon, @pfalcon"
o;-)

dpgeorge, 2017-01-30:
@kfricke I checked the list of users who are blocked from the micropython organisation, but there is no one there.  And I see you can still open issues (and I assume PRs) and comment, etc.  So is it just that you don't get @pfalcon's notifications?

kfricke, 2017-01-30:
Yes, he did block me personally in his Github profile. This way i do not get any notification of his activity on Github (including his activities on MicroPyhton, esptool, etc.).

Sadly i do have few ways to let him know and talk with him about it.

dpgeorge, 2017-01-31:
OK, well I can't do much about it, sorry!

kfricke, 2017-01-31:
I know, thanks anyways.

pfalcon, 2017-05-06:
Not a bug report, closing.

closed 2017-05-06 by pfalcon