micropython/micropython #2768: REPL serial speed

fratics, 2017-01-07:
Hello, I'm tryng to install MP ESP8266 on Cactus Micro Rev.2 but I think there are some problems to use REPL at 115200 baud. I want to know if it's possible to build a MP version (I have sdk and build env) with the serial speed of REPL at 38400 baud.
Thank you in advance!
fratics

robert-hh, 2017-01-07:
esp_mphal.c, line 47
change UART_BIT_RATE_115200 to UART_BIT_RATE_38400

fratics, 2017-01-07:
YES, I already make this change here but the speed appear to be 112500 baud...
**void mp_hal_init(void) {
    //ets_wdt_disable(); // it's a pain while developing
    mp_hal_rtc_init();
    uart_init(UART_BIT_RATE_115200, UART_BIT_RATE_115200);
}**
I rebuild after a make clean but the speed appear not change.

fratics

robert-hh, 2017-01-07:
Before posting, I tried that, and it worked. I tried it again to be sure. The function now looks like:
```
void mp_hal_init(void) {
    //ets_wdt_disable(); // it's a pain while developing
    mp_hal_rtc_init();
    uart_init(UART_BIT_RATE_38400, UART_BIT_RATE_38400);
}
```
Did you deploy the new image after making it with
`make deploy`

fratics, 2017-01-07:
Perfect! now it's working! :)
fratics

robert-hh, 2017-01-07:
Good. Don't forget to close the issue.

closed 2017-01-07 by fratics