micropython/micropython #2769: ESP8266: type object 'socket' has not attribute 'timeout'

jeancf, 2017-01-07:
I am running micropython 1.8.6 on a Wemos D1 mini v1.

I have an application that uses UDP sockets. It runs but I have a problem catching `timeout` exceptions. This code:

    from machine import UART
    import socket

    uart = UART(0, 115200)

    s = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
    s.settimeout(2.0)
    s.bind(("", 9999))
    try:
        s.receive(10)
    except socket.timeout:
        uart.write("Exception caught\n")

throws:

    Traceback (most recent call last):
      File "main.py", line 12, in <module>
    AttributeError: type object 'socket' has no attribute 'timeout'

Instead of [catching the exception](http://docs.micropython.org/en/latest/esp8266/library/usocket.html#usocket.socket.settimeout).

pfalcon, 2017-01-07:
Please see https://github.com/micropython/micropython/commit/a1a8f01799c7a638db52ae008e85143c4b3f215a

closed 2017-01-07 by pfalcon

jeancf, 2017-01-07:
Thanks. Will the docstring automatically update the documentation on http://docs.micropython.org ?

pfalcon, 2017-01-07:
Yes, update happens every night, so will be there tomorrow.

jeancf, 2017-01-07:
Lovely, thanks.

elsasskiting, 2017-01-17:
Bonsoir Jean, 
Je me permets de poster un message sur ce sujet pour rentrer en contact avec vous.
Je voudrai exporter un forum hébergé chez forumactif. 
J'ai suivi à la lettre les recommandations de Roromis. Tous c'est bien passer lors de l'exportation jusqu'à 93 % pour un soucis de format UFTH à cause d'un pseudo d'un membre qui utilisait des symboles spéciaux.
J'ai modifié son pseudo dans le forum et dans le fichier save.pickle 
Puis j'ai  relancé le script en gardant le fichier save.pickle  pour le relancer là ou il a buggé.  
Mais le script s’arrête directement à 0%. 
Du coup j'ai supprimé le fichier save.pickle pour recommencer entièrement l'export.
Et là il bloque déjà à 39 %. 
En comparent les fichiers save.pickle du premier et deuxième export : ils ne sont pas identiques comme si la configuration du forumactif aurait été modifiée par les gens qui gèrent forumactif. 
J'ai fait des tests hier soir pendant 5 heures , seraient-ils si réactifs pour modifier un paramètre pour empêcher   toute export ? 
Du coup je me retrouve avec un fichier save.pickle du premier export, dans lequel je pense avoir récupéré tous les sujets et messages, mais qui est inutilisable pour mes faibles connaissances en informatiques ( quelques notions en php, html et sql ) . Existe-il un script pour utiliser directement ce fichier pour import les donnée dans un nouvelle BD ? 
Merci d'avance. 
On est une petite association de kitesurf qui voudrait intégrer l'ancien forum ( très riche en sujets et messages ) dans notre nouveau site wordpress
http://kitealsace.forumactif.com/

elsasskiting, 2017-01-18:
Bonsoir,  c'est bon j'ai trouvé le problème, un message que j'ai supprimé qui posait problème. 

Merci d'avoir aidé à développer ce script qui nous rend un grand service.

jeancf, 2017-01-18:
Pas de souci. Bonne chance pour la suite.

/~JC

-------- Message original --------
Objet : Re: [micropython/micropython] ESP8266: type object 'socket' has not attribute 'timeout' (#2769)