micropython/micropython #2772: Installing external python modules

elcapwne, 2017-01-08:
Hello everyone!

Is it possible to install external modules (other than the ones micropython supports)? i.e scapy for packet crafting.

Thank you.

peterhinch, 2017-01-10:
This is a FAQ and should be raised in the forum rather than in the bug tracker. I suggest you read [this](https://github.com/micropython/micropython/wiki/Differences).

pfalcon, 2017-01-10:
Another good response would be: if you're interested whether module X works with MicroPython, please ask authors/maintainers of that module, and let them know that you'd like it to run on MicroPython.

For example I, as the author of https://github.com/pfalcon/picotui module can say: yes, it works with MicroPython!

closed 2017-01-10 by pfalcon