micropython/micropython #2773: ESP8866 - 'priority' and 'wake' params in Pin.irq  method are not supported

rsc1975, 2017-01-08:
The doc (v1.8.6 and v1.8.7) seems to be wrong for the port esp8266, the parameters `priority` and `wake` are not supported, but they appears in the `Pin class` doc: https://micropython.org/resources/docs/en/latest/esp8266/library/machine.Pin.html

I've seen in the port Pin implementation that those parameters are not supported: https://github.com/micropython/micropython/blob/master/esp8266/machine_pin.c

Besides, there are some constants, like `Pin.IRQ_LOW_LEVEL` or `Pin.IRQ_HIGH_LEVEL` that are not in Pin class but they appears in the doc.

It seems that there are some other issues related with doc customizations by port, maybe that are sphinx issues, but I don't think so.

BugsForDays, 2018-07-20:
I have also noticed this issue on v1.9.4

yunqingshaw, 2019-09-18:
we have the same problem with no Pin.IRQ_LOW_LEVEL or Pin.IRQ_HIGH_LEVEL. do you have any suggestion?

jimmo, 2019-09-19:
@yunqingshaw these parameters are only supported on the CC3200 (WiPy) port. ESP8266 doesn't (to my knowledge) provide the functionality of `IRQ_LOW_LEVEL` (i.e. trigger the IRQ if the pin is held low for a specified number of ticks).

Can you use `IRQ_FALLING` or `IRQ_RISING` instead?

But the overall issue described in this bug (the docs don't clearly explain the port-specific differences) is an area that could use some improvement. Since this bug was originally raised, the port-specific docs have been combined into common docs, but there's still some work to go to clarify all the port-specific bits.

robert-hh, 2019-09-19:
Indeed, the 'new' structure of the documentation is confusing. People always stumble over features that are not present in their respective board. And that single hint at the start of the libraries section is easy to missl.

byronczimmer, 2022-08-23:
IRQ_LOW_LEVEL and IRQ_HIGH_LEVEL are also not available on the rp2/pico - and it took a while to finally interrogate dir(Pin) to discover they are not available.

The ability to detect a 'long press' natively would be fantastic and help with using a limited number of buttons as input.