micropython/micropython #2774: py: optimise 3-argument power for large integers

dpgeorge, 2017-01-09:
The 3-argument form of the builtin "pow" function could do with optimisation when the arguments are large integers.  The use-case is cryptography.  See https://github.com/micropython/micropython/issues/682#issuecomment-270948125 and subsequent comments for previous discussion.

dpgeorge, 2017-01-09:
The relevant mpz function is already written, just commented out.  I did some simple tests with this function and got the following results for the existing unoptimised code (numbers chosen so it would run in reasonable time on a PC):
```
calculation: pow(x, y, z)

x = 3 << 17
z = 1234123412341234567

  y    time (s)  RAM (bytes)
---------------------------------
1<<15     0.45      229651
1<<16     1.6       458023
1<<17     6.3       914767
1<<18    24.8      1828255
1<<19    99.2      3655231
```
When doubling "y" the time is going up x4 and the RAM up x2, which agrees with the theoretical big-O, as mentioned by @nickovs in the previous discussion.

Using the optimised function the above tests all finish in <0.03s with <3k RAM usage.

dmazzella, 2017-01-21:
@dpgeorge where i can find optimised function?

dpgeorge, 2017-01-22:
@dmazzella grep for mpz_pow3_inpl

dpgeorge, 2018-12-04:
The optimised pow3 implementation was enabled for use in df0117c8ae213a0652c3b19a969edc7fd994eeab

closed 2018-12-04 by dpgeorge