micropython/micropython #2775: ESP8266 - Board auto-reset when calling machine.time_pulse_us()

rsc1975, 2017-01-09:
I'm using a ESP8266 port (v1.8.7 and v1.8.6), The board is a Wemos D1 mini PRO.

When I call the method 'machine.time_pulse_us()' using the default timeout or a lower value the method works perfectly, but If I use higher timeouts (3 seconds or more) then the board is automatically reset, the timeout error is not raised.

In my tests the GPIO doesn't change, so the timeout should be raised.

```python
>>> import machine
>>> machine.time_pulse_us(machine.Pin(0), 1, 1000000)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 110] ETIMEDOUT
>>> machine.time_pulse_us(machine.Pin(0), 1, 1000000)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 110] ETIMEDOUT
>>> machine.time_pulse_us(machine.Pin(0), 1, 1000000)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 110] ETIMEDOUT
>>> machine.time_pulse_us(machine.Pin(0), 1, 3000000)

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 32028, room 16 
tail 12
chksum 0x40
ho 0 tail 12 room 4
load 0x3ffe8000, len 1092, room 12 
tail 8
chksum 0x17
load 0x3ffe8450, len 3000, room 0 
tail 8
chksum 0x56
csum 0x56
l��|��rrnb��l�b�lb쌜����b�lrlll��|��rrnb��ll�pr�b�b쌜��b��lrl�l��|��rrnb��l���b�b쌜���b�lbl����n�r��n|�llll`��r�l�l�l`��r�l�l�l`��r�l���ll`rl��rl���b��b�bbr�rb��n�nn�l��l�l��ll������l�n����bll�rp���bl�brlrlr�n����brb��b쎜����lllb�b��n����brb��b쎜����lllbl�l��|��rrnb��l��b�b쌜��|�bllbl��ln�l`��ln�prl�l��|��rrnb���#5 ets_task(40100164, 3, 3fff8398, 4)
could not open file 'main.py' for reading

MicroPython v1.8.7-7-gb5a1a20a3 on 2017-01-09; ESP module with ESP8266
Type "help()" for more information.
>>> 
```

I don't know if this is a issue related with my board or with the core. Is there some timeout limit value for this method ? 

BTW, I'm not sure if this can be relevant, in my board the "ticks limit" is 1073741823.

```python
>>> import time
>>> time.ticks_add(0, -1)
1073741823
```

dpgeorge, 2017-01-10:
I can confirm the issue.  It looks like the WDT is resetting the board because the busy-wait loop in time_pulse_us (waiting for the edge on the pin) does not feed the watch dog.

It's difficult to solve this problem.  The time_pulse_us function should be accurate to the microsecond (or close to it), but if you need to wait more than a couple of seconds for the edges then background wifi processing needs to happen while waiting.  If the edge changes during the background processing then your accuracy will be off.

rsc1975, 2017-01-10:
Hi @dpgeorge, I discover this issue in a use case that can be achieved using a different approach, I need to wait for a Pin pulse (PIR sensor), but actually I didn't need the time until the event.

I was used the method with an ultrasonic sensor and It worked perfectly, so I thought  that It could be a good alternative to code like this (waiting activation of a PIR sensor):

```python
while True: 
    reading = pin0.value()
    if reading == 1:
        # Do something
    time.sleep(1)
```

I think that, for my PIR sensor, the best approach is keep the board in sleep mode and wake it up when the pulse is sent to the pin, unfortunately the esp8266 port doesn't support it (wake from pin pulse), so I suppose that I'll use a classic polling approach.

Anyway, If we cannot use longer timeouts in the method `time_pulse_us()`, I suppose that It should be commented in the docs, and use it only for use cases with relatively short timeouts (i.e ultrasonic sensor). It would be nice an alternative, not so accurate, maybe in seconds, for long waiting scenarios.

dpgeorge, 2017-01-11:
> I suppose that I'll use a classic polling approach or interrupts with Pin.irq().

Interrupts on falling/rising edge would seem like the best way to do it.

> Anyway, If we cannot use longer timeouts in the method time_pulse_us(), I suppose that It should be commented in the docs,

It is a limitation of the esp8266 port, not a general limitation.  Either the esp8266 has such a limitation on short times, and has accurate resolution, or it has no limitation but has reduced resolution.

rsc1975, 2017-01-11:
if the esp8266 port has some limitations or peculiarities on the method `time_pulse_us()`, I would include them in the docs, but It seems that there are some other issues with the docs customizations per port, I opened another issue related with the esp8266 port docs:  #2773

Is this a "work in progress" ? I mean adapt the doc of the common modules and classes to the specific port features if not, can we (the community) help with that ?

goatchurchprime, 2019-06-22:
I've just encountered this and had to slowly work it out for myself before finding it here.  

Would it be helpful if the esp8266 version threw an exception when given a number that was too large, just so programmers can discover the issue before wasting time chasing after the wrong idea (I thought it was to do with faulty wiring)?

dpgeorge, 2019-06-24:
See #4868 for a fix for this issue.

dpgeorge, 2019-07-01:
Fixed by 89a23a05b3feb2f81fb00d272202b274c2a325cd

closed 2019-07-01 by dpgeorge