micropython/micropython #2776: esp8226: select.poll() UART compatibility

Pablode, 2017-01-09:
Trying to use select.poll to forward data between UART and a wifi connection.
Initializing, reading and writing to UART does work.

According to the MicroPython ESP8266 docs, [objects that support polling are: pyb.UART, pyb.USB_VCP](http://docs.micropython.org/en/latest/esp8266/library/select.html).
On ESP8266, this is [not properly implemented yet](https://github.com/micropython/micropython/blob/master/esp8266/machine_uart.c#L258).

This example code does not work:

```python
import select
from machine import UART

uart = UART(0, 115200)
handler = select.poll()
handler.register(uart, select.POLLIN)
handler.poll()
```
`OSError: [Errno 22] EINVAL`

Would love to see this fixed.

Pablode, 2017-01-16:
```c
STATIC mp_uint_t pyb_uart_ioctl(mp_obj_t self_in, mp_uint_t request, mp_uint_t flags, int *errcode) {
    mp_uint_t ret;
    if (request == MP_STREAM_POLL) {
        ret = 0;

        if ((flags & MP_STREAM_POLL_RD) && uart_rx_char() != -1) {
            ret |= MP_STREAM_POLL_RD;
        }
        if ((flags & MP_STREAM_POLL_WR) && true/*TODO: check if we can write*/) {
            ret |= MP_STREAM_POLL_WR;
        }

    } else {
        *errcode = MP_EINVAL;
        ret = MP_STREAM_ERROR;
    }
    return ret;
}
```

Partially implemented the required function.

Pablode, 2017-01-22:
Please have a look. I haven't tested it.

[`machine_uart.c`](https://github.com/micropython/micropython/blob/master/esp8266/machine_uart.c#L257) (line 257)
```C
STATIC mp_uint_t pyb_uart_ioctl(mp_obj_t self_in, mp_uint_t request, mp_uint_t flags, int *errcode) {
    mp_uint_t ret;
    if (request == MP_STREAM_POLL) {
        ret = 0;
        if ((flags & MP_STREAM_POLL_RD) && uart_rx_has_char()) {
            ret |= MP_STREAM_POLL_RD;
        }
        if ((flags & MP_STREAM_POLL_WR) && uart_tx_can_write()) {
            ret |= MP_STREAM_POLL_WR;
        }
    } else {
        *errcode = MP_EINVAL;
        ret = MP_STREAM_ERROR;
    }
    return ret;
}
```

[`uart.h`](https://github.com/micropython/micropython/blob/master/esp8266/uart.h#L95) (line 95)
```C
bool uart_rx_has_char(void);
bool uart_tx_can_write(uint8 uart);
```

[`uart.c`](https://github.com/micropython/micropython/blob/master/esp8266/uart.c#L202)
```C
bool uart_rx_has_char(void) {
    return input_buf.iget != input_buf.iput;
}
bool uart_tx_can_write(uint8 uart) {
    uint32 fifo_cnt = READ_PERI_REG(UART_STATUS(uart)) & (UART_TXFIFO_CNT<<UART_TXFIFO_CNT_S);
    return (fifo_cnt >> UART_TXFIFO_CNT_S & UART_TXFIFO_CNT) < 126;
}
```

dpgeorge, 2017-02-03:
Now implemented.

closed 2017-02-03 by dpgeorge