micropython/micropython #2778: WPA2 Enterprise support?

udupa-varun, 2017-01-09:
The network security protocols that can be handled are listed as WEP, WPA-PSK and WPA2-PSK. Are there any plans to support WPA2-Enterprise PEAP in the near future? Would love to see that included.

udupa-varun, 2017-01-10:
Of course, sorry about that. I should have mentioned that I meant the ESP8266 NodeMCU development boards. For example, the Adafruit HUZZAH ESP8266 Breakout or the Lolin NODEMCU v3.

wilatai, 2017-02-02:
If possible, I would love to see this supported for ESP8266 NodeMCU boards.

It would open so many possibilities in business and education environments that frequently use this form of network security.

pfalcon, 2017-02-02:
> It would open so many possibilities in business and education environments that frequently use this form of network security.

Feel free to sponsor this development.

wilatai, 2017-02-03:
What are the options to sponsor this development? 
Is there a donation/bounty/crowdfunding system in place somewhere?

pfalcon, 2017-02-03:
There's a universal bounty platform for Open Source project, you can watch an attempt to use it in this ticket: https://github.com/micropython/micropython/issues/888#issuecomment-232808482

For serious ("business and education") offers please contact George Robotics, the creators of MicroPython directly using the information at the site: http://micropython.org/

wilatai, 2017-02-03:
Thank you for your reply. I'll try to check it out in the near future!

Although I can see the possibilities of WPA2 enterprise support for business and education (for example the frequently mentioned and sought after `Eduroam` support) world wide, I can only consider a contribution on a personal basis. Academic organizations are (in my experience) unfortunately, often not that flexible when it comes to funding/sponsoring these kind of initiatives. (Everything that needs a credit card, PayPal, etc. has to be paid in advance by faculty members themselves.. and requires approval of N people. ;-o ).

Hopefully a system like that one you mentioned could help bring a number of people together to sponsor this functionality. :)

hickford, 2017-02-28:
I'm also interested in connecting to enterprise wi-fi (eap? peap?) from esp8266  

    ssid, username, password

hickford, 2017-03-09:
I've been reading to try understand how WPA2 Enterprise could be implemented.

This looks to be the relevant micropython code https://github.com/micropython/micropython/blob/master/esp8266/modnetwork.c#L97

It calls functions (viz. `wifi_station_set_config` and `wifi_station_connect`) in the Espressif esp8266 SDK  https://espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf

There's a pertinent section in the SDK documentation about WPA2 Enterprise

> ### 3.14. WPA2_Enterprise APIs
> 
> ESP8266 Station can connect to WPA2_Enterprise APs.
> WPA2_Enterprise APIs can be found in /ESP8266_NONOS_SDK/include/
> wpa2_enterprise.h
> 
> #### 3.14.15.WPA2_Enterprise Work Flow
> 
> Here is the work flow that prepares ESP266 station to connect to WPA2_Enterprise AP.
> 
> 1. Call wifi_station_set_config to set the configuration of target AP.
> 1. Call wifi_station_set_wpa2_enterprise_auth(1); to enable WPA2_Enterprise authentication.
>    * For EAP-TLS authentication, call wifi_station_set_enterprise_cert_key to set certificate and private key. wifi_station_set_enterprise_username is an optional choice, it can be called to set user name.
>    * For EAP-TTLS or EAP-PEAP authentication, call wifi_station_set_enterprise_username and wifi_station_set_enterprise_password to set user name and password. wifi_station_set_enterprise_ca_cert is an optional choice, it can be called to set root certificate.
> 1. Call wifi_station_connect to connect to target AP.
> 1. After being connected to an AP, or failing to connect to AP and on stopped retries, please call the corresponding wifi_station_clear_enterprise_XXX APIs to release the resources

kfricke, 2017-03-09:
The real problem is not the client side to implement this. Try to find an enterprise environment where you can implement, thoroughly test and play with this feature.
Of course one can implement such an infrastructure with Linux based components (all components are available as opensource tools), but the amount of work to create and play with such an infrastructure is bigger than embedding some methods from the SDK into MicroPython's network API.

Jostikas, 2017-05-03:
A first go at trying to "embed some methods" resulted in overflowing the irom0_0_seg, as libwpa2 and libcrypto seem to both be needed.

hickford, 2018-03-08:
I see that Pycom document ["Connecting to a WPA2-Enterprise network"](https://docs.pycom.io/chapter/tutorials/all/wlan.html#connecting-to-a-wpa2-enterprise-network). Is this useful to us? Can we copy code from their fork?

joostd, 2018-03-08:
Probably, but note that Pycom produces esp32 based boards, so [the micropython implementation for these boards](https://github.com/pycom/pycom-micropython-sigfox) is based on [espressif's esp-idf](https://github.com/espressif/esp-idf) targeting the esp32, and using a [closed source wifi library](https://github.com/espressif/esp32-wifi-lib) also specific to the esp32. Pycom seems to have [ported to the ESP8266](https://github.com/pycom/pycom-micropython-sigfox/tree/master/esp8266) as well, but I'm not sure if that supports wpa2-enterprise?

dpgeorge, 2018-03-09:
> I see that Pycom document "Connecting to a WPA2-Enterprise network". Is this useful to us? Can we copy code from their fork?

No, please do not copy any code from there because their license is not compatible with the MIT license used here.

hickford, 2018-03-09:
@joostd I have both a esp8266 Wemos D1 mini and a esp32 Wemos lolin32.

@dpgeorge Thanks Damien for your reply. Shucks. I'd hoped the [collaboration Pycom announced last year](https://forum.pycom.io/topic/550/pycom-and-damien-george-join-forces-for-the-esp32) would make that possible.

> We are pleased to confirm that after further discussions with Damien George, we have decided to work together on a single Port of MicroPython for the ESP32. This Port will be MIT licensed thus enabling Pycom and the existing MP community to build a super-rich and quality code base hosted on the official MicroPython repository.

hickford, 2018-03-15:
Good news maybe? https://github.com/espressif/esp32-wifi-lib/issues/2#issuecomment-361554740

> Parts of WPA2 and WPS source code are planned to be open sourced in [Espressif IDF] 3.1.

Abd0s, 2019-08-29:
Is this now implemented in the lib? I remember reading it in a changelog, but can't find anything about it in docs etc,

dpgeorge, 2019-08-30:
> Is this now implemented in the lib?

No.

phatcatmatt, 2019-11-21:
> I'm also interested in connecting to enterprise wi-fi 802.1x EAP (ssid, username, password) from micropython on esp8266

Plus one from me.

FalconBFR, 2021-01-28:
+1 - my school uses WPA2 Enterprise. How can I help contribute to this? I have no experience with contributing to Micropython but I have some basic understanding of Arduino and C code.

SlitheringMule, 2021-05-23:
I'd love to see this feature added.

qermit, 2021-07-11:
Hi. I started modifying the esp8266 port to enable WPA Enterprise but I have some problems in linking phase

```Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
LINK build-GENERIC/firmware.elf
build-GENERIC/modnetwork.o:(.text.esp_connect+0x14): undefined reference to `wifi_station_set_wpa2_enterprise_auth'
build-GENERIC/modnetwork.o:(.text.esp_connect+0x18): undefined reference to `wifi_station_set_enterprise_identity'
build-GENERIC/modnetwork.o:(.text.esp_connect+0x1c): undefined reference to `wifi_station_set_enterprise_username'
build-GENERIC/modnetwork.o:(.text.esp_connect+0x20): undefined reference to `wifi_station_set_enterprise_password'
build-GENERIC/modnetwork.o: In function `esp_connect':
modnetwork.c:(.text.esp_connect+0x185): undefined reference to `wifi_station_set_wpa2_enterprise_auth'
modnetwork.c:(.text.esp_connect+0x19a): undefined reference to `wifi_station_set_enterprise_identity'
modnetwork.c:(.text.esp_connect+0x1b4): undefined reference to `wifi_station_set_enterprise_username'
modnetwork.c:(.text.esp_connect+0x1c6): undefined reference to `wifi_station_set_enterprise_password'
make: *** [Makefile:217: build-GENERIC/firmware.elf] Error 1```

When I add additional libraries i get following error:
```Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
LINK build-GENERIC/firmware.elf
/home/qermit/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libwpa.a(sha256-internal.o): In function `sha256_init':
(.text.sha256_init+0x20): multiple definition of `sha256_init'
build-GENERIC/extmod/moduhashlib.o:moduhashlib.c:(.text.sha256_init+0x28): first defined here
make: *** [Makefile:217: build-GENERIC/firmware.elf] Error 1```

qermit, 2021-07-13:
I have found `sha256_init` definition in `extmod/crypto-algorithms/sha256.c`

tonyallan, 2022-05-10:
> Has anyone achieved this yet?

No I haven't but at the time this was an issue I arranged for the organisation to create a restricted 2.5GHz WPA2 Wi-FI network for IoT.

MaxMyzer, 2023-01-19:
I came across this while looking for a way to get an RP2 to connect, I suppose its gonna be a while

tigercoding56, 2023-01-26:
is there w way to connect to wpa enterprise in normal python ? , if so maybe someone could port relevant code