micropython/micropython #2779: Should there be machine.bootloader() ?

pfalcon, 2017-01-10:
My current connection can't even load https://github.com/micropython/micropython/issues/1430 , so opening separate ticket. I see that machine module docs don't describe `machine.bootloader()` (as evolution of `pyb.bootloader()`). Should it be there or do we want to implement it adhoc from port to port?

robert-hh, 2017-01-10:
For devices like the huzzah feather, where esptool switches to boot-loader state  by flipping DTR and RTS, this seems not to be urgently needed. For other devices it's helpful.

dpgeorge, 2017-01-16:
I find it very useful for stmhal-based boards.  I use it all the time and only revert to using the BOOT0 pin if there's a problem with the firmware and I can't use machine.bootloader().

On some boards there is no bootloader (eg microbit, where an external chip is responsible for flashing the firmware) but on boards where there is one it seems like machine.bootloader() is a sensible thing to have.  It's similar to having machine.reset().

dpgeorge, 2021-05-01:
Docs for this added in dd5c831a0b0c960b638109f1886edc10178f51a1

closed 2021-05-01 by dpgeorge