micropython/micropython #2781: Add ssl client certificate on ESP8266

Lir10, 2017-01-11:
Hi , 

I'm trying to connect to ssl server using this code : 

```
>>>import usocket                                                                                                       
>>> import ussl                                                                                                          
>>> s = usocket.socket(usocket.AF_INET, usocket.SOCK_STREAM)                                                             
>>> ss = ussl.wrap_socket(s,  ca_certs='/flash/cert/ca.pem')     

```
However , im getting the following tranceback :           
                                  
```
Traceback (most recent call last):                                                                                       
  File "<stdin>", line 1, in <module>                                                                                    
TypeError: extra keyword arguments given 
```

Looks like ussl.wrap_socket can get only one argument.

How can i add my own self signed certificate + client certificate to the ssl socket object?

hiway, 2017-01-11:
> Warning: Currently, this function does NOT validate server certificates, which makes an SSL connection established prone to man-in-the-middle attacks.

https://docs.micropython.org/en/latest/esp8266/library/ussl.html

This will work:

    >>> ss = ussl.wrap_socket(s)

But as the docs warn, it is not validated.

Lir10, 2017-01-11:
Thanks for the answer , 

I wonder what's the point to make ssl connection without certificates ? 
So what's the difference between that and regular tcp connection ??

briant-spindance, 2017-09-06:
Just wondering if there are any plans to resolve this issue?

dpgeorge, 2017-09-06:
> Just wondering if there are any plans to resolve this issue?

The axtls library used on esp8266 does support certificates (although I'm not sure to what extent).  So it's possible to add support for them but it's not planned at this stage.  An alternative way to add certs (and at the same time getting more advanced crypto algorithms) would be to support mbedtls on the esp8266, but the esp8266 might not have enough RAM for this.

pfalcon, 2017-09-06:
There're definitely plans to implement that, it's just plans for hundreds of other things have more priority, and implementing hundreds of things takes years, so nobody yet got to it. @Lir10 , @briant-spindance , you seem to need this more than other folks, so feel free to prepare patches for it. (High quality ones, which generally goes without saying. Please start at https://github.com/micropython/micropython/blob/master/CONTRIBUTING.md before doing any coding work.)

> would be to support mbedtls on the esp8266, but the esp8266 might not have enough RAM for this.

axTLS was originally selected as a default TLS implementation (even before esp8266 port) because it has minimal code size and RAM requirements.

Vrettel, 2017-12-08:
Hello,

I recently tested SSL/TLS connection on my ESP8266 with micropython. The test was done with self-signed certificates from the server-broker that I passed to my ESP. Using the umqtt (simple.py) I found out as mentioned in an earlier post that the function ussl.wrap_socket() works only with one argument. With only one argument I can't include my certificate (ca_certs). By giving only the socket (first argument) the connections is successful. I tried to capture some packets and the connection seems to be encrypted. 
My question is; If anyone can connect without certificate but the connection is encrypted, how secure is this connection ? 

Thanks in advance

peterhinch, 2017-12-09:
See the docs [here](http://docs.micropython.org/en/latest/esp8266/esp8266/general.html?highlight=tls) and [here](http://docs.micropython.org/en/latest/esp8266/library/ussl.html).

dpgeorge, 2017-12-13:
> My question is; If anyone can connect without certificate but the connection is encrypted, how secure is this connection ?

You'll need to implement *authentication* yourself.  Eg using a password with umqtt.

Vrettel, 2017-12-13:
@peter and @dpgeorge  thank you for your replies. I 'll keep my connection secure with username/password and the current encryption method.

Sunrise17, 2018-02-09:
@dpgeorge , i am using simple umqtt for establishing SSL connection to AWS IoT with ESP32 board, but i have not succeeded to establish connection. I am getting following error from ussl.wrap_socket method;
`Error in mqtt connect: [Exception] TypeError: extra keyword arguments given
`
self.params equals = {'keyfile': '/flash/cert/deviceCert.key', 'certfile': '/flash/cert/deviceCertAndCACert.pem', 'ca_certs': '/flash/cert/root.pem'}

Connect Method in umqtt is seen below, is it possible to establish full cert chain connection to AWS IoT?
`def connect(self, clean_session=True):
        self.sock = socket.socket()
        addr = socket.getaddrinfo(self.server, self.port)[0][-1]
        self.sock.connect(addr)
        print(self.ssl_params)
        if self.ssl:
            import ussl
            self.sock = ussl.wrap_socket(self.sock, **self.ssl_params)
...`

Sunrise17, 2018-05-12:
In micropython version 1.9.4, has this issue been resolved?

peterhinch, 2018-05-13:
It seems that it has. See [here](https://github.com/micropython/micropython/releases/tag/v1.9.4) (scroll down to extmod section, fourth line reads "modussl_axtls: implement key and cert kw args to wrap_socket".

Sunrise17, 2018-05-13:
I am trying to connect esp32 to AWS IoT broker (i have tried both ports 8883 and 443 regarding to latest [update](https://aws.amazon.com/about-aws/whats-new/2018/02/aws-iot-core-now-supports-mqtt-connections-with-certificate-based-client-authentication-on-port-443/) of AWS),

Connect function in mqtt.py;
[image]
 
but i am receiving following error;
[image]

dpgeorge, 2018-05-17:
@Sunrise17 the esp32 (and esp8266) supports the parameters `key` and `cert` when creating an SSL connection.  Note that these are not the same as `keyfile` and `certfile`.  The parameters `key` and `cert` need to be passed actual key/cert data, not a filename.  You can load a file like this:
```python
with open(keyfile, 'rb') as f:
    key = f.read()
```
That should work.  Note that the `ca_cert` parameter is currently not implemented.

Sunrise17, 2018-05-17:
Dear @dpgeorge , thanks for your response, i have tried your method to connect aws not but still i am receiving same error. As you will see that ssl params passed as data in following below;

[image]

In AWS IoT documentation, format to publish mqtt message as follows;

    $ mosquitto_pub --cafile root.cert --cert deviceCertAndCACert.crt --key deviceCert.key -h <prefix>.iot.us-east-1.amazonaws.com -p 8883 -q 1 -t  foo/bar -i  anyclientID --tls-version tlsv1.2 -m "Hello" -d

So,  enabling cacert is necessary i think to establish connection direct connection from esp32 to AWS IoT?

dpgeorge, 2018-05-18:
@Sunrise17 you shouldn't post your private keys like that!  I suggest you change them immediately.

It's hard to debug this problem without extra info.  What is line 65 of mqtt.py?

Sunrise17, 2018-05-18:
@dpgeorge , please find attacked mqtt.py file. Line 65 is : 
    self.sock.connect(addr)

[mqtt.zip](https://github.com/micropython/micropython/files/2015500/mqtt.zip)

Connect line in main.py is: 

    CACERT_PATH = "/flash/cert/root.pem"
    KEY_PATH = "/flash/cert/deviceCert.key"
    CERT_PATH = "/flash/cert/deviceCertAndCACert.pem"

    with open(KEY_PATH, 'rb') as f:
        key1 = f.read()

    with open(CERT_PATH, 'rb') as f:
        cert1 = f.read()

    client = MQTTClient(client_id=MQTT_CLIENT_ID, server=MQTT_HOST, port=MQTT_PORT, keepalive=10000, ssl=True, ssl_params={ "key":key1, "cert":cert1, "server_side":True }) #, "ca_certs":CACERT_PATH})

dpgeorge, 2018-05-18:
> please find attacked mqtt.py file. Line 65 is :
> self.sock.connect(addr)

If it's failing on this line then it doesn't have anything to do with SSL (SSL handshake happens after this line succeeds).  If it fails here then there is something wrong with the internet connection and/or the remote server (maybe it's domain name is wrong?).

peterhinch, 2018-05-18:
I think someone should write a FAQ on using SSL/TLS as this issue keeps cropping up. I'd offer but a) it's not my field and b) I never managed to get it to work either ;-)

Sunrise17, 2018-05-31:
Dear @dpgeorge ,

I have tried three option for line 48 for path versions of key and cert file in wrap_socket method and i am receiving following errors in each try as below. (Note: key1 and cert1 returns bytes object, in option 3 that's why i have converted it to string and try) With same cert and key files, i have tried Python 3.5 from PC, it is well connecting to server but i could not succeed to connect from esp32 board.

    KEY_PATH = "/flash/cert/client.key"
    CERT_PATH = "/flash/cert/client.crt"
    HOST, PORT = "45.255.66.124", 5800

    with open(KEY_PATH, 'rb') as f:
        key1 = f.read()

    with open(CERT_PATH, 'rb') as f:
        cert1 = f.read()

    45 ::      s = usocket.socket(usocket.AF_INET, usocket.SOCK_STREAM)
    46 ::      addr = usocket.getaddrinfo(HOST, PORT)[0][-1]
    47 ::      s.connect(addr)

**Option_1**
    48 ::      sock = ussl.wrap_socket(addr, key = KEY_PATH, cert = CERT_PATH)

Error: (gives error and continuously resetting the board)
    assertion "ret == 0" failed: file "../../extmod/modussl_mbedtls.c", line 182, function: socket_new
    abort() was called at PC 0x401368e7 on core 0

    Backtrace: 0x4008c983:0x3ffb6530 0x4008c9af:0x3ffb6550 0x401368e7:0x3ffb6570     0x400ed05d:0x3ffb65a0 0x400dfb5a:0x3ffb65f0 0x400dbe49:0x3ffb6620 0x400dbeb1:0x3ffb6640 0x400e8f0d:0x3ffb6660 0x400dfbe8:0x3ffb6700 0x400dbe49:0x3ffb6780 0x400e8e81:0x3ffb67a0 0x400dfbe8:0x3ffb6840 0x400dbe49:0x3ffb68c0 0x400dbe76:0x3ffb68e0 0x400f872a:0x3ffb6900 0x400f8981:0x3ffb69a0 0x400ef907:0x3ffb69c0

    Rebooting...
    Guru Meditation Error: Core  0 panic'ed ()
    Guru Meditation Error: Core  0 panic'ed (Guru Meditation Error: Core  0 panic'ed ()
    Guru Meditation Error: Core  0 panic'ed ()
    Guru Meditation Error: Core  0 panic'ed (Guru Meditation Error: Core  0 panic'ed (Guru Meditation Error: Core  0 panic'ed (Guru Meditation Error: Core  0 panic'ed (Unknown reason)
ets Jun  8 2016 00:22:57

**Option_2**
    48 ::      sock = ussl.wrap_socket(addr, key = key1, cert = cert1)

Error:
    File "main.py", line 48, in DownloadFile
    OSError: stream operation not supported

**Option_3**
    48 ::      sock = ussl.wrap_socket(addr, key = key1.decode("utf-8"), cert = cert1.decode("utf-8"))

Error:
    File "main.py", line 48, in DownloadFile
    OSError: stream operation not supported

dpgeorge, 2018-06-01:
@Sunrise17 you need to pass `s` to `wrap_socket`, not `addr`, like this:
```python
sock = ussl.wrap_socket(s, key=key1, cert=cert1)
```

Sunrise17, 2018-06-01:
@dpgeorge , thanks for your warning, it works well right now between SSL Python Server and ESP32 Client. In other side, to connect AWS IoT over mqtt with ESP32 board, i am still trying to solve it. Amazon is requiring ca_cert parameter in connection may be that's why its not connecting. After i have found some solution, i will write it here. Thanks again...

StanS-AWS, 2019-01-06:
I can connect to AWS MQTT with client cert using ESP32 WROOM-32 module. But the same exactly code and certs fail to connect when running on ESP8266 board. The error code is "invalid key". Both flashed micropython 1.9.4, the same exactly certs, but different results. Is there is a limit on cert length in 8266 build? What other factors may affect the connection?

Swington, 2019-01-08:
@StanS-AWS I guess the difference is in SDKs for ESP8266 and ESP32, cause there is actually a difference between ssl libs in C for those two platforms.
I have exactly the same problem and according to ESP8266 manual, it's SDK provides full ssl support. So I deduce that this functionality just isn't translated into Micropython API.

StanS-AWS, 2019-01-13:
It turned-out ESP8266 certs need to be in .der format to work correctly. The code remains the same.
[](https://awsiot.wordpress.com/2019/01/10/connect-8266-to-aws-mqtt-using-miropython/)

StanS-AWS, 2019-01-13:
https://awsiot.wordpress.com/2019/01/10/connect-8266-to-aws-mqtt-using-miropython/

SlowBro904, 2020-02-12:
Does validation work properly on ESP32?

jrichrd, 2020-10-06:
> Note that the `ca_cert` parameter is currently not implemented.

Does anyone know if the 'ca_cert' parameter is now implemented in the latest version of MicroPython? I'm not able to get it to work using the code samples above however I can connect if I remove the 'ca_cert' and use only the key and device cert.

LMtx, 2020-10-15:
> Does anyone know if the 'ca_cert' parameter is now implemented in the latest version of MicroPython? I'm not able to get it to work using the code samples above however I can connect if I remove the [...]

I would love to know this as well. But it looks like 'ca_cert' is still not implemented :( That is a big issue for any production-like usage of this library.

jimmo, 2020-10-19:
`ca_certs` is currently only implemented in the CC3200 port (WiPy). See https://github.com/micropython/micropython/pull/5998 for an ESP32 implementation.

Donatussss, 2024-05-13:
Hello, I'm currently facing the `invalid_key` error on `ESP8266` as well as `Raspberry Pi Pico W` even after converting the files to `.der` format using `openssl`. Anyone else facing this issue recently? Will really appreciate your help.

Saranya-karan, 2024-08-26:
> Hello, I'm currently facing the `invalid_key` error on `ESP8266` as well as `Raspberry Pi Pico W` even after converting the files to `.der` format using `openssl`. Anyone else facing this issue recently? Will really appreciate your help.

hi iam also facing the error while using pico w..my error code is
Traceback (most recent call last):
  File "<stdin>", line 72, in <module>
  File "simple1.py", line 84, in connect
IndexError: bytes index out of range

do u find any solution?

Donatussss, 2024-08-26:
Hi @Saranya-karan please view [this comment](https://github.com/aws-samples/aws-iot-core-getting-started-micropython/issues/1#issuecomment-2254986537) I made. You'll tell me if it worked

Saranya-karan, 2024-08-26:
[simple1.txt](https://github.com/user-attachments/files/16750760/simple1.txt)
i attched the simple1.py in  the text format sir.hope u can view this

Donatussss, 2024-08-26:
Okay, please send a screenshot of the error

Saranya-karan, 2024-08-26:
[image]

Donatussss, 2024-08-26:
Could be your connection details are invalid cause you're not getting a response from the broker

Saranya-karan, 2024-08-26:
because of my network ?sir can you elaborate the issue.

Saranya-karan, 2024-08-26:
> Could be your connection details are invalid cause you're not getting a response from the broker

sir can you elaborat the issue

Saranya-karan, 2024-08-26:
> Could be your connection details are invalid cause you're not getting a response from the broker

with the same certificates and end point i sent data to aws iot core in arduino ide(ESP32 board).but in micropython(picow board) it is making issues

Donatussss, 2024-08-26:
I see, have you tried decoding the base64 text as I pointed out in [that other comment](https://github.com/aws-samples/aws-iot-core-getting-started-micropython/issues/1#issuecomment-2254986537)

Donatussss, 2024-08-26:
To do the base64 decoding you need the certificates in base64 format which is the certificate.pem.crt and private.pem.key files

Saranya-karan, 2024-08-26:
> I see, have you tried decoding the base64 text as I pointed out in [that other comment](https://github.com/aws-samples/aws-iot-core-getting-started-micropython/issues/1#issuecomment-2254986537)

sir may i know which version of micropython u used for sending data to aws..
iam using 1.20.0

Donatussss, 2024-08-26:
I am pretty much using latest version

Saranya-karan, 2024-08-26:
> To do the base64 decoding you need the certificates in base64 format which is the certificate.pem.crt and private.pem.key files

[image]
```
def read_cert(filename):        
    with open(filename, 'rb') as f:
        text = f.read().strip()
        split_text = text.split(b'\n')  # Split using bytes newline
        base64_text = b''.join(split_text[1:-1])  # Join the byte strings
        return ubinascii.a2b_base64(base64_text)

DEV_KEY = read_cert('/d5126a8b49f0e6423dd6f5129d8a493feaf53fbf393e3355d3532ca988a51978-private.pem.key')
DEV_CRT = read_cert('/d5126a8b49f0e6423dd6f5129d8a493feaf53fbf393e3355d3532ca988a51978-certificate.pem.crt')
yes i tried that way also.still finding the same issue

Donatussss, 2024-08-26:
Okay, I'm not sure I can offer much help using comments, is it possible that you have your project in a repo then I can clone it and try on the pico w I have. This I'll do tomorrow

Saranya-karan, 2024-08-26:
> Okay, I'm not sure I can offer much help using comments, is it possible that you have your project in a repo then I can clone it and try on the pico w I have. You can have it private then add me as a collaborator. I'll work on it tomorrow

I will add the repo tomorrow sir. I provided my simple1. py library and main program. Nothing much there. thank you so much for the respose

Saranya-karan, 2024-08-27:
> Okay, I'm not sure I can offer much help using comments, is it possible that you have your project in a repo then I can clone it and try on the pico w I have. You can have it private then add me as a collaborator. I'll work on it tomorrow

i added the files to my repo..it is in public view.pls let me know u can access the files..

Saranya-karan, 2024-08-27:
> Okay, I'm not sure I can offer much help using comments, is it possible that you have your project in a repo then I can clone it and try on the pico w I have. You can have it private then add me as a collaborator. I'll work on it tomorrow

sir any feedbacks?

Donatussss, 2024-08-28:
Hi @Saranya-karan , I've made changes to the repo. The code seems to work on my end. [image] [image].
You'll try it out and let me know. Also note the version of micropython.
I've also made it such that you don't have to hardcode your credentials everytime you create a new project needing the credentials. This will also be helpful when provisioning many devices and also when sharing your project

jonnor, 2024-09-29:
The original problem (as well as several other, semi-related) problems seems to have been fixed. Perhaps we can close this issue? For usage questions or debugging help, the recommended is to open a Github Discussion or come in the MicroPython Discord.