micropython/micropython #2783: retrieve long long int (int64_t) from mp_obj_int

abiliojr, 2017-01-11:
I want to obtain a 64 bit number from an arguments, very much like:
```
seconds = mp_obj_get_int(args[0]);
```

But this only works for 32 bits. Is there any way to do this in a direct way?

stinos, 2017-01-11:
This is not really a guthub issue, better ask at the forums http://forum.micropython.org/.
Or have a look at how it's done here: https://github.com/stinos/micropython-wrap/blob/master/detail/frompyobj.h#L173 note there's some ifdefs to deal with differences between building 32bit/64bit

pfalcon, 2017-02-10:
> I want to obtain a 64 bit number from an arguments, very much like:
> seconds = mp_obj_get_int(args[0]);

On 64-bit platform, you will obtain what you want. And on 32-bit platform you can't rely that there will be 64 bits of int at all. (This is explanation why we don't have what you want in a single call and probably won't have.) And if you rely that uPy is built with MPZ support (which doesn't have to be the case!), you can use MPZ API to get to it as shown by @stinos .

closed 2017-02-10 by pfalcon