micropython/micropython #2784: Is there a convinent way to access string/bytes/bytearray/list item?

nupha, 2017-01-12:
Is there function or macro make it easy to access element from an object. Like methods __getitem__, __setitem__  in CPython. Prototypes may look like:

```
mp_obj_t mp_obj_setitem(mp_obj_t dest, mp_uint_t idx, mp_obj_t val);
mp_obj_getitem(mp_obj_t dest, mp_uint_t idx);
```

It could be better to make __setitem__, __getitem__  generic methods that could be overriden.

nupha, 2017-01-12:
Fix, that should be \_\_getitem\_\_, \_\_setitem\_\_ in CPython

nupha, 2017-01-12:
Also, variants are considered:

```
mp_obj_setitem_int(mp_obj_t dest, mp_uint_t idx, mp_int_t val);
mp_int_t mp_obj_getitem_int(mp_obj_t dest, mp_uint_t idx);

mp_obj_setitem(mp_obj_t dest, mp_uint_t idx, byte abyte);
byte mp_obj_getitem_byte(mp_obj_t dest, mp_uint_t idx);

...
```

dhylands, 2017-01-12:
[mp_obj_get_array](https://github.com/micropython/micropython/blob/master/py/obj.h#L666) can be used on tuples and lists.

[mp_get_buffer](https://github.com/micropython/micropython/blob/master/py/obj.h#L666) can be used to get at the data for a bytestring/bytearray.

nupha, 2017-01-12:
Thanks.
I'm using `mp_get_buffer_raise` now, but I suggest a more generic way for customization.

dpgeorge, 2017-01-12:
mp_obj_subscr(base, index, value) does generic subscription on all types.  Use valuel=MP_OBJ_NULL for delete, value=MP_OBJ_SENTINEL for load, any other value to store.

closed 2017-01-17 by dpgeorge