micropython/micropython #2785: Method for getting RSSI of the currently connected AP?

mika-koivusaari, 2017-01-12:
Currently there doesn't seem to be a method for getting rssi for currently connected ap. WLAN.scan does return rssi, but it takes some time and also return lots of unnecessary information.

I propose adding a method for getting the current rssi.

As a proof of concept I added such a method https://github.com/mika-koivusaari/micropython/tree/esp8266-rssi

But it would most probably be best to discuss before even making a pull-request. Questions that come mind are at least.
1. What name should the method have?
2.Should it have some standard range it returns, or should it just return whatever the underlying hw returns?
3. Error values? Exceptions would be more pythonic than some magic value?

Comments?

dpgeorge, 2017-01-16:
Thanks for the initiative.  Other contributors may chime in here, but otherwise please see #2602.

kfricke, 2017-02-06:
From the first glimpse this method makes only sense in the station mode interface. So that check should be added at least.

This method will most likely not make it into a general network API, because it is only valid for wireless networks.
So there should be another place i would guess. Like an additional `ifconfig()` method as an read-only `rssi` argument. On Unix the old-style `ifconfig` command also returns lower network layer statictics.

mika-koivusaari, 2017-02-06:
I think network.WLAN class would be a obvious place.

kfricke, 2017-02-06:
Of course.

kfricke, 2017-02-10:
...but adding an additional new method is more impact on the API than adding another keyword argument to a method which is used in a similar environment.
Even with @pfalcon's call for an RFC regarding unifying/redesigning the network API, this is of least impact and can be seen as a way to not further deface it.

dpgeorge, 2018-02-26:
As discussed in #3351, `WLAN.status('rssi')` is now the way to query the signal of the AP that the STA is connected to.  And in c5fe610ba15468e1d92d7b6d5f5962f7595e3324 this was implemented for esp8266.

closed 2018-02-26 by dpgeorge

Informaticore, 2021-02-03:
@dpgeorge if I am using this I always get the same value. Am I using it wrong? 

```
wifi = network.WLAN(network.STA_IF)
wifi.active(True)
wifi.connect(***, ***)
...
wifi.status('rssi')
```

I am actually calling this wifi.status('rssi') in an asyncio task every 3 seconds and it always prints the same value, no matter if I move the device or the AP. If I reboot and get a fresh connection the value changes but then stays exactly the same.