micropython/micropython #2786: stmhal: Problem with __lshrdi3 (libgcc link issue?)

prusnak, 2017-01-12:
We hit a strange issues, which we think are caused by some weird configuration of linker.

Steps to reproduce:

1) add the following to `stmhal/main.c`:

``` C
--- a/stmhal/main.c
+++ b/stmhal/main.c
@@ -146,7 +146,14 @@
     }
 }
 
+extern uint64_t __lshrdi3(uint64_t a, uint64_t b);
+
 int main(void) {
+
+    __lshrdi3(3, 4);
+
+    return 1;
+
     // TODO disable JTAG
 
     // Stack limit should be less than real stack size, so we have a chance

```

2) uncomment the following line in `stmhal/Makefile`:

``` Makefile
LIBS += $(shell $(CC) -print-libgcc-file-name)  
```

3) compile and upload to device

4) device will enter HardFault

prusnak, 2017-01-12:
For reference, this is the code which emits call to `__lshrdi3`:

``` C
    volatile int s = 37;
    volatile uint32_t a = 0xFFFFFFFF, b = 0xFFFFFFFF, c = 0xFFFFFFFF;
    uint64_t res;
    res = (a + (uint64_t)b * c) >> s;
```

prusnak, 2017-01-12:
Debugging further in assembler it seems that `__lshrdi3` function jumps into some weird location on `+2` line:

```
Dump of assembler code for function __lshrdi3:
=> 0x0803b138 <+0>:	adds	r0, #32
   0x0803b13a <+2>:	b.n	0x803b5e2
   0x0803b13c <+4>:	stmia	r0!, {r5}
   0x0803b13e <+6>:	b.n	0x803b606
   0x0803b140 <+8>:	lsls	r0, r6, #8
   0x0803b142 <+10>:	sbcs	r0, r4
   0x0803b144 <+12>:	lsls	r1, r6, #12
   0x0803b146 <+14>:	str	r0, [r4, r6]
   0x0803b148 <+16>:	lsrs	r1, r2, #16
   0x0803b14a <+18>:	sbcs	r0, r0
   0x0803b14c <+20>:	asrs	r1, r6, #8
   0x0803b14e <+22>:	b.n	0x803b492
   0x0803b150 <+24>:	vrhadd.u16	d14, d14, d31
```

dhylands, 2017-01-12:
I'd try correcting the call to include libgcc. Change it from:
```
LIBS += $(shell $(CC) -print-libgcc-file-name)
```
to be:
```
LIBS += $(shell $(CC) $(CFLAGS) -print-libgcc-file-name)
```
The compiler comes with a bunch of different libgcc's and CFLAGS is needed to tell it to select the correct one.

prusnak, 2017-01-12:
@dhylands CORRECT! Thank you very much. Sent fix in https://github.com/micropython/micropython/pull/2787

closed 2017-01-12 by prusnak