micropython/micropython #2789: unix REPL paste ignores extended ASCII characters

peterhinch, 2017-01-13:
Copy the following and paste it at the REPL:
```python
z = 'ÇØĚ'
```
In paste mode it works, but not at the REPL prompt where the string contents disappears.

dpgeorge, 2017-01-16:
The interactive REPL does not support unicode/utf8.  It requires some effort to implement because one needs to support backspace/delete of utf8-encoded chars, as well as moving left/right across them.

peterhinch, 2017-01-16:
Fair enough - but might this cause problems for non-English speakers?
Is simply swallowing the characters the best way to deal with it? I encountered this when debugging a program. A long string contained a few instances of chr(176) - the degree symbol. The REPL let me paste the string while silently modifying it. It took me a while to spot that it was the REPL rather than my code that was the culprit.

dpgeorge, 2017-01-17:
> but might this cause problems for non-English speakers?

Yes.  We support utf8 in strings so should support it at the REPL as well.  The simplest thing to do is just allow through all bytes (greater than 126 in value) to the REPL line.  Beyond that, supporting left/right cursor motion, and deletion, is more tricky.

laowu2019, 2020-03-04:
Is there any plan to implement this unicode support for REPL?

As far as i know, it will be very helpful to chinese users.

Sdator, 2023-05-20:
This function is needed very much, which can make us use repl more amicably. I hope it can be put on the agenda as soon as possible. Thank the team for their efforts.

jonnor, 2024-09-29:
Tested on MicroPython 1.24 with Unix port. Issue still present - the characters do not appear.