micropython/micropython #2792: Timer period unable to set to microseconds

ccspoz, 2017-01-15:
Is there a reason why timer periods are measured in milliseconds while the timers on the pyboard are in microseconds? Is there a way to get sub-ms granularity with the timers? I am trying to implement AC phase control with a triac and the millisecond limitation is essentially limiting my output to only 10 discrete steps.

Apologies if the answer can be found here already, or if this is a newbie question- I have been searching all day to no avail.

dhylands, 2017-01-15:
This question is more appropriate for the forum.

What board are you using? The pyboard times can go all the way to 1/42 MHz granularity.

Can you provide some code? Here's some code for the pyboard that uses microsecond granularity:
https://github.com/dhylands/upy-examples/blob/master/ic_test.py

ccspoz, 2017-01-16:
Oh, my bad, will use the forum in the future, thanks.

I am trying to use the ESP12e board.

The code I am attempting to use is:

from machine import Timer
tim = Timer(-1) 
tim.Init(period=2, mode=Timer.ONE_SHOT, callback=timer_callback)

And this virtual timer works but with only ms granularity. I realise the timing at these levels is not accurate but being able to set in us would provide for smooth fading.

Is it currently possible to implement hardware timers on the esp8266?

pfalcon, 2017-05-06:
Thanks for switching to forum. Closing.

closed 2017-05-06 by pfalcon