micropython/micropython #2793: Blocking socket read ignores Ctrl+C

jeremyherbert, 2017-01-16:
Not sure if this is known or not, but I figured I would report it anyway. The `umqtt.simple.MQTTClient` in micropython-lib has a function `wait_msg` continues to block even through ctrl+c. This is because it uses a blocking socket read. I'm not sure if this behaviour of socket is intended, but in terms of consistency I'm pretty sure that CPython allows you to Ctrl+C out of blocking reads.

pfalcon, 2017-01-20:
No, it's not known. Please start with specifying your platform. There may be more useful hints at http://www.chiark.greenend.org.uk/~sgtatham/bugs.html .

jeremyherbert, 2017-01-20:
I am using the lopy, perhaps this is just an issue in their implementation of wifi based sockets?

pfalcon, 2017-01-20:
Very much likely, because all mainline MicroPython targets are carefully implemented to support Ctrl+C for socket operations. You're recommended to follow LoPy vendor's recommendations for support (i.e. their own forum), or if by some reason you don't want to do that, there may be a section for Pycom's products at http://forum.micropython.org . (But the official way is still to follow Pycom's process.)

jeremyherbert, 2017-01-20:
Ok no problems. Thank you!

closed 2017-01-20 by jeremyherbert