micropython/micropython #2800: WebREPL crashes when trying to connect

d33tah, 2017-01-18:
```
WebREPL connection from: ('192.168.0.140', 38040)
Traceback (most recent call last):
  File "webrepl.py", line 40, in accept_conn
  File "websocket_helper.py", line 27, in server_handshake
ValueError: need more than 1 values to unpack
Traceback (most recent call last):
  File "webrepl.py", line 40, in accept_conn
  File "websocket_helper.py", line 23, in server_handshake
OSError: EOF in headers
Traceback (most recent call last):
  File "webrepl.py", line 40, in accept_conn
  File "websocket_helper.py", line 34, in server_handshake
OSError: Not a websocket request
```

Tested on Wemos D1 mini, latest master.

dpgeorge, 2017-01-19:
The latest master works ok for me.  What webrepl client version are you using and what browser?

d33tah, 2017-02-07:
Perhaps it's worth giving a hint in the error message then?

d33tah, 2017-02-07:
BTW @zaphodddd don't edit your post next time, I almost missed it. If you wrote another one I'd get a notification ;)

zaphodddd, 2017-02-07:
That'd be quite helpful. Sorry for the comment edit  - I was a bit ashamed after I figured out what happened  :)

pfalcon, 2017-04-05:
> Perhaps it's worth giving a hint in the error message then?

Feel free to elaborate what you mean here.

In the meantime: the fact that you're getting an error should already give enough hints that you're doing something wrong and it's time to read the docs ;-). A computer is generally not smart enough to give more *useful* hints beyond that.

d33tah, 2017-04-05:
@pfalcon I meant an error message that points the user towards "you did something wrong, RTFM" as opposed to "it might be the case that you just hit a bug".

jonnor, 2024-09-29:
Problem seems to have been resolved.

closed 2024-09-29 by jonnor