micropython/micropython #2806: appveyor failing on builtin_help test

dpgeorge, 2017-01-23:
Since 1864f90e9a2c0cc12e91c88ddd491b809c5317e5 it seems that appveyol is failing on the builtin_help test.  I'd say it should be skipped automatically, because there is no built-in help function on the windows build, but something is not working.

@stinos can you please look into this when you have some time, thanks.

stinos, 2017-01-23:
The standard unix build has the same problem, the output is `######## FAIL` so seems to be some problem with handling special tests?

stinos, 2017-01-23:
Two problems here:
- there's no support fro skipping the special tests, easily fixed by adding a `if output_mupy == b'SKIP\n': return output_mupy` at line 146, I'll add a PR for that
- treating tests as special relies directly on the arguments passed in by the user so calling `./run-tests ./basics/builtin_help.py` is not the same as calling `./run-tests basics/builtin_help.py`

commit 533129f8 (2017-01-23): tests: Make sure special tests can be skipped as well.

Fixes #2806.
closed 2017-01-23 by dpgeorge