micropython/micropython #2812: micropython (unix) binary has executable stack

SpotlightKid, 2017-01-27:
I'm building a [AUR package](https://aur.archlinux.org/packages/micropython-git/) of micropython (git version) for Arch Linux:

The package checking tool (namcap) warns me that the `micropython` program has an executable stack (an thus would not work on hardened kernels). This can be easily confirmed:

``` 
$ readelf -l micropython | grep -A 1 GNU_STACK
  GNU_STACK      0x0000000000000000 0x0000000000000000 0x0000000000000000
                 0x0000000000000000 0x0000000000000000  RWE    10
```
(Should be `RW` not `RWE`.)

It can be fixed by adding `-Wa,--noexecstack` to CFLAGS:

    make CFLAGS_EXTRA="-Wa,--noexecstack"

I see no adverse affect of disabling the executable flag on the stack. All tests still pass:

```
$ make CFLAGS_EXTRA="-Wa,--noexecstack" test
[...]
570 tests performed (18154 individual testcases)
570 tests passed
11 tests skipped: builtin_help cmd_parsetree framebuf1 framebuf16 framebuf4 urandom_extra vfs_fat_fileio vfs_fat_oldproto vfs_fat_ramdisk buffered_writer extra_coverage
```

Should these flags be added by default?

dpgeorge, 2017-01-30:
The code does generate its own executable machine code, but that's placed in special executable memory, not the stack, so I'd say it's OK to make the stack non-executable.

But how do other programs deal with this problem for the AUR?  I doubt many packages use the `--noexecstack` option so they'd all have the same issue.

dhylands, 2017-01-30:
If you add the following line:
```
    .section    .note.GNU-stack,"",@progbits
```
to the very end (after the #endiif) of the .S files in the .py directory then the stack of micropython will be non-executable.

If the linker doesn't find a .note.GNU-stack section in an input file then it assumes that the stack is marked as executable (this is largely for historical reasons).

There are occaisons where an executable stack is needed. See: https://www.win.tue.nl/~aeb/linux/hh/protection.html but these trampoline functions are not needed in micropython.

dpgeorge, 2017-01-30:
@SpotlightKid please let us know if @dhylands suggestion fixes the problem for you.

SpotlightKid, 2017-01-30:
Indeed it does.

Here's the patch:

```
diff --git a/py/nlrx64.S b/py/nlrx64.S
index caea35de2..2572a2538 100644
--- a/py/nlrx64.S
+++ b/py/nlrx64.S
@@ -257,3 +257,5 @@ nlr_jump:
 #endif // !defined(NLR_OS_WINDOWS)
 
 #endif // defined(__x86_64__) && !MICROPY_NLR_SETJMP
+
+.section    .note.GNU-stack,"",@progbits
diff --git a/py/nlrx86.S b/py/nlrx86.S
index 8c538ba17..a1b4e9529 100644
--- a/py/nlrx86.S
+++ b/py/nlrx86.S
@@ -190,3 +190,5 @@ nlr_jump:
 #endif
 
 #endif // defined(__i386__) && !MICROPY_NLR_SETJMP
+
+.section    .note.GNU-stack,"",@progbits
diff --git a/py/nlrxtensa.S b/py/nlrxtensa.S
index 289996cce..0e9976509 100644
--- a/py/nlrxtensa.S
+++ b/py/nlrxtensa.S
@@ -114,3 +114,5 @@ nlr_jump:
     .size   nlr_jump, .-nlr_jump
 
 #endif // defined(__xtensa__)
+
+.section    .note.GNU-stack,"",@progbits
```

dpgeorge, 2017-01-30:
Will there be a problem compiling this on other archs/platform/compilers?

pfalcon, 2017-01-30:
I'd definitely err on that side ;-)

dhylands, 2017-01-30:
The asm files are already gcc specific.

The directive is arch independent.
The directive  needs to be outside the #if otherwise you're back to having a .o with no .note.GNU-stack section which causes the stack of the final elf to be marked executable.

dpgeorge, 2017-01-31:
Ok, maybe someone can make a PR, then others can test it, then we can merge if all is good.

dhylands, 2017-01-31:
I needed to wrap the .stack directive in an #if defined(linux). The arm-none-eabi toolchain doesn't like the syntax. I verified that gcc on the RPi was fine with the syntax (so the exeutable stack is not an x86 specific thing).

It passed travis tests except for a missing repository related to coveralls which I don't think is related to this PR.

pfalcon, 2017-02-01:
This should be fixed now.

closed 2017-02-01 by pfalcon