micropython/micropython #2813: py/objtype: Compatibility enabling power special method

AaronKel, 2017-01-27:
Hello all,
I needed to overload the power operator so I added in:
`
    [MP_BINARY_OP_POWER] = MP_QSTR___pow__,
`
`
    [MP_BINARY_OP_INPLACE_POWER] = MP_QSTR___iadd__,
`
This works great for my code but in general if something shouldn't have a power overload it gives the following error
`
TypeError: unsupported types for __pow__: 'tuple', 'int'
`
Instead of:
`
TypeError: unsupported types for **: 'tuple', 'int'
`
I had a little look in py/runtime.c at line 560 but it seems to just take in 
`
mp_binary_op_method_name[op]
`
and from there I am at a loss because in objtype.c they all follow a similar convention in terms of the [op] value and searching to see if "MP_QSTR___add__" for example is defined anywhere else draws a blank.

dpgeorge, 2017-01-30:
> This works great for my code but in general if something shouldn't have a power overload it gives the following error
> TypeError: unsupported types for __pow__: 'tuple', 'int'
> Instead of:
> TypeError: unsupported types for **: 'tuple', 'int'

This is the intended behaviour.  In order to not unnecessarily use up code space, the table that is used to lookup the method name (the dunder method names) is the same table that is used to print the error message.  If you want to customise it to print nicer errors messages then you'll need to add a second table (see eg mp_binary_op_method_name in py/objtype.c).

AaronKel, 2017-01-30:
Ah! Makes sense, thanks for clarifying!

closed 2017-01-30 by AaronKel