micropython/micropython #2815: Unable to request from Timer

mleonard87, 2017-01-28:
I am attempting to make a request to a webserver with in a method that is run on a `Timer`. If I run the method and request outside of the timer it works fine, however, when run inside of a timer I never seem to get a response from the server. The code below illustrates this.

I have also tried constructing the get request with `socket` as per the method found here https://docs.micropython.org/en/latest/esp8266/esp8266/tutorial/network_tcp.html#http-get-request and I still get the same issue.

This is when running on an ESP8266, specifically a NodeMCU.

Any help would be much appreciated.

Thanks, Mike

```python
=== import urequests
=== from machine import Timer

=== def make_request():
===     print('Making request')
===     resp = urequests.get('http://immense-thicket-90803.herokuapp.com/now?tz=Australia/Sydney')
===    print(resp.content)
===
=== make_request()
===
=== tim = Timer(-1)
=== tim.init(period=10000, mode=Timer.PERIODIC, callback=lambda t: make_request())
===
Making request
b'5:34\n'
>>> Making request
Traceback (most recent call last):
  File "<stdin>", line 12, in <lambda>
  File "<stdin>", line 6, in make_request
  File "urequests.py", line 101, in get
  File "urequests.py", line 77, in request
ValueError: need more than 0 values to unpack
Making request
Traceback (most recent call last):
  File "<stdin>", line 12, in <lambda>
  File "<stdin>", line 6, in make_request
  File "urequests.py", line 101, in get
  File "urequests.py", line 77, in request
ValueError: need more than 0 values to unpack
```

peterhinch, 2017-01-28:
Support requests should be raised on the [forum](http://forum.micropython.org/) rather than here; this is a bug tracker. There is discussion of the issue of what can and can't be done in an ESP8266 timer callback [here](http://forum.micropython.org/viewtopic.php?f=16&t=2497&p=14663&hilit=timer#p14663). The upshot is that socket operations aren't guaranteed to work.

pfalcon, 2017-01-30:
Docs are now explicit of requirements was *any* callback as used by `machine`: http://docs.micropython.org/en/latest/esp8266/library/machine.html http://docs.micropython.org/en/latest/esp8266/library/machine.Timer.html

closed 2017-01-30 by pfalcon