micropython/micropython #2817: object's __hash__

cfaessler, 2017-01-29:
As far as I can see, there is no member __hash__ in the class object. What is the reason for this?

dpgeorge, 2017-01-30:
What do you mean by "class object"?

The builtin hash() function works on all builtin types, and also defers to `__hash__` for user defined classes, eg this works:
```python
class A:
    def __hash__(self):
        return 123
hash(A())
```

AaronKel, 2017-02-02:
@dpgeorge I see that floats don't have a hash, why is this? I had a look around to see what was already implemented in CPython and in the decimals library I saw that they have a hash for decimals that produces the same hash as a float. They use a tuple consisting of
`
        return hash((self._sign,
                     self._exp+len(self._int),
                     self._int.rstrip('0')))
`
in which I'm guessing are represented as strings. This isn't true for python 3 onward however as I believe hashes are randomized on startup with a seed to prevent against DoS attacks.

dpgeorge, 2017-02-03:
> I see that floats don't have a hash, why is this?

Yes you are right, neither float nor complex are hashable.  The reason is that no-one has needed it before.

AaronKel, 2017-02-03:
I encountered it as a bug when trying to compare tuples to check if a float is contained in it, I just set the hash to be return a new small int that gives a random value and the error went away. I'm guessing that it checks for value equivalence when using the 'in' keyword?

dpgeorge, 2017-02-03:
> I'm guessing that it checks for value equivalence when using the 'in' keyword?

Yes.  As long as the hash is deterministic (the same value for the object returns the same hash) then everything should work.

pfalcon, 2017-02-03:
@nickovs : For your perspective. Over 3 years, with thousands of users (again, estimated based on stars) nobody tried to to use a float number as a key in dictionary. What to say about 3-arg pow and other obscure stuff - most people even don't know that it exists!

I think that we should fix this issue ASAP. But given that it took so many years to notice, and it's still only 1 person, my hand doesn't raise to set it is priority high, only medium. And when I say "fix this", I mean float, not complex. The latter would need to wait for its own real-world report from some physicist. And as a physicist would never put a complex as a key into dictionary, such report would never happen.

dpgeorge, 2017-02-03:
> And when I say "fix this", I mean float, not complex. The latter would need to wait for its own real-world report from some physicist

I actually got such a request a few weeks ago to fix this exact problem, for both float and complex.

dbrgn, 2017-02-06:
The [`flufl.enum`](https://github.com/dbrgn/RPLCD/blob/master/RPLCD/enum.py) implementation (backport of enums) actually tries to access `object.__hash__`: https://github.com/dbrgn/RPLCD/blob/master/RPLCD/enum.py#L215

Does that mean that enums using this library are not currently possible?

cfaessler, 2017-02-06:
thx @dbrgn thats exactly why I need this :) 
Since the only members the enum has are name and value, we could just implement the `__hash__` function with something like `hash(name+value)` instead of using `object.__hash__`

AaronKel, 2017-02-06:
@cfaessler You would need to hash it as a tuple with the various objects.

pfalcon, 2017-04-05:
Fixed in https://github.com/micropython/micropython/commit/19f2e47d59b539f94a49137c379c88f0aed53cf1, closing.

closed 2017-04-05 by pfalcon