micropython/micropython #2818: VFAT problems

kamikaze, 2017-01-29:
Looks like commit 84c614e [breaks PyBoard`s factory reset](https://forum.micropython.org/viewtopic.php?f=6&t=2968)

And also I've started to have a problem with `import display` since that commit. You can try it yourself by copying "display" directory into fs and then do `import display`. Hint: it's about importing other modules inside display. This code works fine before this breaking commit.

FIY: I'm holding my files on SD card (fsck is ok).

dhylands, 2017-01-29:
I can confirm that factory reset works on commit 3242cf2 and doesn't work on 84c614e. Tested on my pyboard v1.1

dpgeorge, 2017-01-30:
Yes, sorry, the filesystem stuff is undergoing some big updates and internal changes.  The factor reset issue should be now fixed by a52635245491214a6fe92dc357ff214bd6ff8892.

dpgeorge, 2017-01-30:
Regarding the issue with "import display": I think it's a bug in gcc with its implementation of alloca and how it does alignment of the pointer returned by alloca.

Looking at the code from py/runtime.c:mp_import_from (on stmhal), the following in an extract of the part of that function that does alloca (the code is `char *dot_name = alloca(dot_name_len)`):
```
 802b918:       f7f5 faff       bl      8020f1a <qstr_len>
 802b91c:       f8d7 8000       ldr.w   r8, [r7]
 802b920:       f108 0401       add.w   r4, r8, #1
 802b924:       eb00 0904       add.w   r9, r0, r4    # r9 now contains the number of bytes to alloca
 802b928:       f109 030e       add.w   r3, r9, #14   # round up...
 802b92c:       f023 0307       bic.w   r3, r3, #7    # ... round r9 up to multiple of 8, put in r3
 802b930:       ebad 0d03       sub.w   sp, sp, r3    # allocate space on stack
 802b934:       466b            mov     r3, sp
 802b936:       ea4f 0ad3       mov.w   sl, r3, lsr #3  # clear ...
 802b93a:       ea4f 06ca       mov.w   r6, sl, lsl #3  # ... lower 3 bits; r6 contains 8-byte aligned pointer
 802b93e:       4642            mov     r2, r8
 802b940:       4659            mov     r1, fp
 802b942:       4630            mov     r0, r6      # the pointer to the stack-allocated RAM
 802b944:       f011 fc31       bl      803d1aa <memcpy>
```
Notice that the alloca'd pointer is further aligned *after* allocating the space on the stack (the part that clears the lower 3 bits).  So what can happen is that if the pointer comes out as 4-byte aligned from the first computation (ptr=sp-len) then it will be decreased below the stack pointer when it is aligned a second time and it then points to unallocated memory on the stack.  I think this is what is happening (based on analysis of the registers and stack contents after a crash, memcpy is coping data over the PC on the stack and returning to an invalid location).

Compare to the code using variable-length-arrays instead of alloca (`char dot_name[dot_name_len]`):
```
 802b918:       f7f5 faff       bl      8020f1a <qstr_len>
 802b91c:       683e            ldr     r6, [r7, #0]
 802b91e:       1c74            adds    r4, r6, #1
 802b920:       eb00 0904       add.w   r9, r0, r4    # r9 now contains the number of bytes to alloca
 802b924:       f109 0307       add.w   r3, r9, #7    # round ...
 802b928:       f023 0307       bic.w   r3, r3, #7    # ... r9 up to multiple of 8
 802b92c:       46e8            mov     r8, sp        # save sp so it can be restored later
 802b92e:       ebad 0d03       sub.w   sp, sp, r3    # allocate space on stack
 802b932:       4632            mov     r2, r6
 802b934:       4651            mov     r1, sl
 802b936:       4668            mov     r0, sp    # the pointer to the stack-allocated RAM
 802b938:       f011 fc33       bl      803d1a2 <memcpy>
```
Here the pointer is aligned only once and sp points exactly to the allocated RAM.  Using this code instead of the previous code makes the crash go away.

@dhylands @pfalcon what do you think?  I don't want to blame the tools but it really does look like gcc has a bug here (confirmed in arm-none-eabi-gcc v5.3.0 and v6.3.0).

I think we should switch to using VLAs instead of alloca.  VLAs are (apparently) part of the C99 standard, whilst alloca is not.

dpgeorge, 2017-01-30:
Continuing from my post above, the alloca implementation is actually OK because sp is guaranteed to be 8-byte aligned on entry to a function, and so it maintains its 8-byte alignment throughout.  Therefore, the additional rounding is a null-operation: it does nothing because the pointer is always 8-byte aligned.

The problem I had was that, using PYBLITE, the stack pointer was only 4-byte aligned on initialisation, which is now fixed by 6b12934fecdf531f9199b835cff1c2dd501d149b.

But still, I'd suggest switching from alloca() to VLA where possible in the code because it generates smaller code and is C99 compliant.

@kamikaze regarding your issue with "import display": I can't reproduce it.  Can you please check if the issue is still there with the latest master, and if so please provide more info as to the problem (board you are using, exact symptom of the issue, etc).

dhylands, 2017-01-30:
In the first example, it adds 14 to the size and rounds that down. So the new size still contains 8 more bytes than needed, which allows for the pointer alignment.

dpgeorge, 2017-01-30:
> In the first example, it adds 14 to the size and rounds that down. So the new size still contains 8 more bytes than needed, which allows for the pointer alignment.

No, the +14 is independent to the second alignment: the `sub.w   sp, sp, r3` instruction allocates the space and from then on sp points to the bottom of the stack.  But then r3/r6 are used to calculate the pointer to the alloca'd object (by further aligning the value of sp), which may point further *below* sp.

dhylands, 2017-01-30:
right - I was "rounding" in the wrong direction. It would need to add the original size to sp and round that in order to be correct, not further round

kamikaze, 2017-01-30:
@dpgeorge yes, even now with latest master I have such problem. But there is a strange thing. I have following code in my main.py:
```
import uav

uav.run_uav_test()
```

from https://github.com/kamikaze/pyboard-examples

which imports display itself and actually works... until i try to enter into REPL. I don't even press CTRL+C and then app hangs and LEDs start to power on incrementally, then poweroff. On third time all of them are on. Game over.

Then I commented out everything in my main.py. Now I can enter into REPL, but still have same problem when trying to import display manually. Or even when I'm trying to execute manually same code from main.py. It hangs right on `import uav`.

I have PyBoard v1.1

kamikaze, 2017-01-30:
Tested on 3242cf2 again - all works fine.

kamikaze, 2017-01-30:
Sorry, actually I'm pressing CTRL+C in REPL

kamikaze, 2017-01-30:
gcc version: gcc-arm-none-eabi-5_4-2016q3

pfalcon, 2017-01-30:
> @pfalcon what do you think? I don't want to blame the tools but it really does look like gcc has a bug here (confirmed in arm-none-eabi-gcc v5.3.0 and v6.3.0).

Wonder I'm still on 4.9.3 for ARM and don't haste to upgrade, seeing enough other people who don't? So, that's the best quick suggestion I'd have - try (use?) a trusty version from a known source, like https://launchpad.net/gcc-arm-embedded/4.9

pfalcon, 2017-01-30:
> the pointer is always 8-byte aligned.

Right, ARM stack is required to be 8-bytes aligned, because of the requirement of load-dword instruction. I was amazed to see that x86-embedded ("iamcu" gcc target) beat ARM on that and uses 4-byte alignment.

> But still, I'd suggest switching from alloca() to VLA where possible in the code because it generates smaller code and is C99 compliant.

There was a reason why VLA used in most places and alloca in some. That's because of a known issue that modern compilers don't respect scoping for making allocations and tend to allocate one big lump for all function's variables at the start of functions. With alloca, they can't. They wouldn't be able with VLAs too, but VLAs is too a novelty, so nobody knows how compilers really make them work. So, someone would need to experiment with different compilers for different archs to see it. If you think you have time for that, feel free to start with putting dynamically differently sized arrays in 2 if branches.

dpgeorge, 2017-01-30:
I fixed a problem with not popping the NLR state when reading a file, see c30b308492ddd587e5a1f76c538956cad44735a2.  @kamikaze that may fix your issue, please test and let me know.

kamikaze, 2017-01-30:
@dpgeorge now it works ok, thanks )

closed 2017-01-30 by kamikaze

dpgeorge, 2017-01-30:
Ok, good that the problem is fixed.

Regarding VLA/alloca: that can be discussed and researched in a different thread, but for now let's just keep it in the back of our minds and do it another day.