micropython/micropython #2820: esp2866 missing poll on uart

mhoffma, 2017-01-30:
UART is missing the poll feature, if you use poll you can configure it but you can't actually issue a poll because the ioctl is defined to return the error condition.

Pablode, 2017-02-01:
Duplicate, see [#2776](https://github.com/micropython/micropython/issues/2776).

mhoffma, 2017-02-01:
thats a great catch. I personally don't care which one gets add to the
repository Looks like the 2776 uses better names, I could put those names
in the patch I'm driving or you can submit 2776.

pfalcon, 2017-02-01:
@Pablode, @mhoffma : To merge something into mainline, there should be a *patch* (on github represented by Pull Request). As #2776 is not a pull request, it can't be merged.  https://github.com/micropython/micropython/pull/2821 is a pull request, but has bugs. Please work together to produce the best patch and test it together. Thanks.

Pablode, 2017-02-02:
Hey. I'm aware that my issue #2776 is no PR. This is, because `bool uart_tx_can_write(uint8 uart)` has not been tested (and I'm not able to). In the project I'm working on, only the reading part is required.

Therefore, go ahead and use my code if you want to. Seems like #2821 is pretty much the same except the function declarations.

mhoffma, 2017-02-02:
I will change the tx function name to what you used uart_tx_can_write if
that works for the group and then hopefully this can get added.

dpgeorge, 2017-02-03:
Now implemented.

closed 2017-02-03 by dpgeorge