micropython/micropython #2825: Build error: make: ../mpy-cross/mpy-cross: No such file or directory

BrendanSimon, 2017-01-30:
The latest code from git does not compile (at least not for stmhal boards such as STM32F4DISCO)

> MPY modules/lcd160cr.py
> make: ../mpy-cross/mpy-cross: No such file or directory
> make: *** [build-STM32F4DISC/frozen_mpy/lcd160cr.mpy] Error 1

dpgeorge, 2017-01-30:
stmhal now uses frozen bytecode by default, so you need to first build the cross-compiler.  This is a recent change and I just now added some instructions to the README to clarify this, see 326343feebdd1ad990864cc830a30b6a44d44918.

closed 2017-01-30 by dpgeorge

BrendanSimon, 2017-02-01:
OK.  That fixes it :)

Is there a way to do this automatically via makefile dependencies?  Or at least warn/notify user to rebuild the cross-compiler if it's out of date?  At the moment I run have to manually build the cross compiler after every git pull before building MPY for my target.