micropython/micropython #2831: Race condition causing SD card mount failure and system hang

ryannathans, 2017-02-01:
Over the last year I have noticed that as the device boots, less and less frequently is the SD card mounted at all. I have been running from flash, checking if the SD is mounted after boot.  

With the recent changes to main.c, my hack to run from flash (stop the chdir to SD and prevent SD being mounted on usb) experienced merge conflicts and such I'm not using it. 

If I put an SD card in my device with latest master, the device hangs on boot every time. It just sits there frozen and never shows up on usb. 

If I put a boatloader before micropython that has around 30 seconds of sleep in it, then the same "hanging" version of micropython works fine. 

As a result I believe that there is a race condition affecting SD card mounting and it has gotten to the point where the SD card is no longer mounted at all under normal conditions. 

 I'm using the STM32F439 and 8gb SanDisk SD, with about 100 manufactured

ryannathans, 2017-02-01:
Putting a 30 second delay on line 271 of main.c also works, going to play with the delay to see what is too short..

ryannathans, 2017-02-01:
HAL_Delay(1000);

Is sufficient to "fix" the problem 😉

dpgeorge, 2017-06-14:
@ryannathans are you still experiencing this issue?  If so can you provide a case that fails on a pyboard?  If not then it's going to be impossible for me to fix, and it could even be an issue with your hardware.

dpgeorge, 2021-04-30:
Closing due to inactivity.  Feel free to reopen with further details on how to reproduce the issue as above.

closed 2021-04-30 by dpgeorge