micropython/micropython #2833: exec('import module') doesn't add module to scope when used in imported file

stinos, 2017-02-01:
Maybe this is intended or known as a difference with CPython but I looked around and didn't see any documentation related to it.

To reproduce (tested on unix-like ports only): create these 3 files:

foo.py:
```
foo = 1
```

bar.py:
```
def Bar():
  exec('import foo')
  print(foo.foo)

if __name__ == "__main__":
  Bar()
```

importbar.py:
```
import bar
bar.Bar()
```

Running `micropython bar.py` works ok and prints `1`.

Running `micropython importbar.py` also imports foo but doesn't add it to the scope somehow:

```
Traceback (most recent call last):
  File "importbar.py", line 2, in <module>
  File "bar.py", line 3, in Bar
NameError: name 'foo' is not defined
```

stinos, 2017-02-02:
Looking into this some more the problem is solely related to the (arguably) wrong scope being used and the 'import' within the exex isn't needed to reproduce the issue. This shows the scope used for exec seems to be the scope which calls the import:

bar.py:
```
def Bar():
  exec('foo=1')
  print('bar', foo)

if __name__ == '__main__':
  Bar()
```

importbar.py:
```
import bar
try:
  bar.Bar()
except:
  print('importbar', foo)
```

result:
```
> micropython bar.py
bar 1
> micropython importbar.py
importbar 1
```

jonnor, 2024-09-29:
Thank you for the investigation and code examples. I tested the first on MicroPython 1.24 Unix port and the behavior seems to be the same. So the issue is still present. However, I wonder in which cases this causes practical problems?

stinos, 2024-09-30:
> However, I wonder in which cases this causes practical problems?

It's not uncommon to use actual Python code for configuring applications, mostly because, well, it's code, so one can do a lot more compared to using .json or .ini or similar for configuration. So I probably ran into this first example when configuring an application by importing such config .py file. However the first example can be worked around simply by using `foo = __import__('foo')` instead of using `exec`.

Apart from that the second example is probably a variation of various other bug reports where micropython treats locals/globals/eval/exec in a different way than CPython does, but I don't think that is ever a real showstopper, it can just cause surprises, but it's always in very specific code which most people never really run into anyway so I'll close this as 'not planned'.

closed 2024-09-30 by stinos