micropython/micropython #2838: Warnings send to stdout instead of stderr

sumpfralle, 2017-02-06:
The following line of code:
```
"foo" == b"foo"
```
causes a warning:
```
Warning: Comparison between bytes and str
```
This is fine.

But surprisingly the warning is sent to stdout instead of stderr:
```
host:~$ micropython -c "'foo' == b''" 2>/dev/null
Warning: Comparison between bytes and str
```
Here I would have expected no output at all.

The current behaviour can break python scripts that are trying to produce a machine readable output.

pfalcon, 2017-02-10:
@sumpfralle : We have 198 config options in MicroPython, and definitely can have 199th to handle this case right. But I'm, as a MicroPython comaintainer, getting a bit weary of adding more and more config options, so for now leaving this for users affected to contribute.

sumpfralle, 2017-02-11:
I understand and would be willing to prepare a changeset.

But (for my understanding): why do you think that this should be a configuration options instead of simply sending all warning messages from the interpreter to stderr instead of stdout?
(I cannot think of a use-case where someone would expect interpreter warnings/errors on stdout)

pfalcon, 2017-02-11:
Because uPy can run on platforms without stderr ;-).

So, you would need to define something like MP_STDERR_PRINTER in py/mpconfig.h, defaulting to what warning.c uses now, and then make unix/mpconfigport.h override it.

sumpfralle, 2017-02-11:
This is indeed quite a good reason. Thank you for pointing that out.

I will take a look at this.

dpgeorge, 2017-09-26:
Fixed in 62849b7010abffb7b0a9c9875930efe7cb77519c

closed 2017-09-26 by dpgeorge