micropython/micropython #2840: ESP8266 documentation for repl connection

blakev, 2017-02-06:
In the [documentation for the repl](https://github.com/micropython/micropython/blob/master/docs/esp8266/tutorial/repl.rst) it has this line for connecting via linux:

```
$ picocom /dev/ttyUSB0
```

However, this is insufficient to get a true interactive session. Instead it should be:

```
$ picocom /dev/ttyUSB0 --b 115200 --omap crcrlf --imap crcrlf
```

kfricke, 2017-02-06:
What do I miss without an additional line feed?

blakev, 2017-02-06:
Ahh you're correct @kfricke ~ however, if the baudrate isn't set it appears like it's not working.

```
$ picocom /dev/ttyUSB0 --b 115200
```

dpgeorge, 2017-02-07:
Thanks for the tip, fixed in 3217bbe4910416b9084691ea5ff1ec1530e15ee4.

closed 2017-02-07 by dpgeorge