micropython/micropython #2842: Compile error

ksgriggs, 2017-02-06:
Compile error while compiling micropython/unix.  I have compiled this before.  Yesterday I ran git pull to get the newest version and now when I run make I get this error.  Any help would be appreciated.  I am running jessie on a rpi2.

Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build/genhdr
Generating build/genhdr/mpversion.h
GEN build/genhdr/qstr.i.last
GEN build/genhdr/qstr.split
GEN build/genhdr/qstrdefs.collected.h
QSTR updated
GEN build/genhdr/qstrdefs.generated.h
CC ../py/mpstate.c
CC ../py/nlrx86.S
../py/nlrx86.S: Assembler messages:
../py/nlrx86.S:194: Error: junk at end of line, first unrecognized character is `,'
../py/mkrules.mk:25: recipe for target 'build/py/nlrx86.o' failed
make: *** [build/py/nlrx86.o] Error 1

dhylands, 2017-02-06:
That's probably related to #2812 

What version of gcc do you have on the RPi?

I'll try building on my RPi (I think its an RPI3) but in the meantime, you can workaround the issue by editing each of the 3 nlr*.S files and removing the last 3 lines of each file:
https://github.com/micropython/micropython/blob/de48a27d607f0e7acbcbd841a8a4c685574114dc/py/nlrx64.S#L260-L262

ksgriggs, 2017-02-07:
gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/4.9/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Raspbian 4.9.2-10' --with-bugurl=file:///usr/share/doc/gcc-4.9/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.9 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.9 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-libitm --disable-libquadmath --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.9-armhf/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.9-armhf --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.9-armhf --with-arch-directory=arm --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 4.9.2 (Raspbian 4.9.2-10)

commit aa34c553 (2017-02-08): py/nlr: Fix execstack builds for ARM.

It seems that the gcc toolchain on the RaspberryPi
likes %progbits instead of @progbits. I verified that
%progbits also works under x86, so this should
fix #2848 and fix #2842

I verified that unix and mpy-cross both compile
on my RaspberryPi and on my x64 machine.
closed 2017-02-08 by dpgeorge